CTSRD-CHERI/cheribsd #1535: Locating capabilities in source code

bond187, 2022-11-01:
I am trying to find in the source code where the ISA extension is described that adds capabilities to the instructions. I'd like to run some small experiments where I change how the checks are done, but I can't figure out where in the source code to look for this. Thank you!

bond187, 2022-11-01:
Thanks so much for your answer. I am new to CHERI and am learning a lot the past couple of days. 

Broadly, I'm trying to design a system that will do a small operation for every read or write type instruction that is performed by the CPU. Eventually it will involve running cryptographic functions for each of these instructions, but for now I'd really like to just do something simple like add two registers and produce the result. This will involve extra overhead, so I want to be able to quantify it by running it in a simulated setting like QEMU, like how CheriBSD is set up. 

That said, I was hoping to clone and modify CheriBSD, since as far as I understand, Cheri ISA does a check of some sort for every instruction. I intended to remove the original check function and replace it with my simple addition scheme and see how much slower that is than without it. Does this seem feasible with CHERI, or should I look elsewhere? Thanks again for your help.

nwf, 2022-11-09:
I believe this discussion was continued [elsewhere](https://www.reddit.com/r/arm/comments/yjfgz9/cheri_isa_extension_modification/) and so this can be closed.

closed 2022-11-09 by nwf