CTSRD-CHERI/cheribsd #1539: Kernel panic in DRM after longer inactivity (panic: running but not TDS_RUNNING)

kwitaszczyk, 2022-11-09:
It happened to me twice that the GENERIC-MORELLO-PURECAP kernel from the dev branch paniced in the DRM code when I logged in a host via SSH to execute some command, with a screen attached and a video console active but not used for some time.

A similar issue I found: https://github.com/evadot/drm-subtree/issues/7.

```
login:   x0: 0xffff00000101b2f0 [rwRW,0xffff00000101b2f0-0xffff00000101b2f8] (vop_open_vp_offsets + 0)                                                                                                  
  x1: 0xffff00019eb33a30 [rwxRW,0xffff00019eb33a30-0xffff00019eb33a50]                                                                                                                                  
  x2: 0xffff00000101b300 [rwRW,0xffff00000101b300-0xffff00000101b350] (vop_open_desc + 0)                                                                                                               
  x3: 0xffff00000101bdf0 [rwRW,0xffff00000101bdf0-0xffff00000101be40] (vop_lock1_desc + 0)                                                                                                              
  x4: 0xffff00000101be50 [rwRW,0xffff00000101be50-0xffff00000101bea0] (vop_unlock_desc + 0)                                                                                                             
  x5: 0xffffa08011d9fd80 [rwxRW,0xffffa08011d9fd80-0xffffa08011d9ff00]                                                                                                                                  
  x6: 0xffff000000f38640 [rwRW,0xffff000000f38640-0xffff000000f386b0] (lock_class_mtx_sleep + 0)                                                                                                        
  x7: 0xffff0000014db4d4 [rwRW,0xffff0000014db480-0xffff0000014db570] (w_locklistdata + 7ffd4)                                                                                                          
  x8: 0x0000000000000010                                                                                                                                                                                
  x9: 0x0000000000000001                                                                                                                                                                                
 x10: 0x0000000000000002                                                                                                                                                                                
 x11: 0x0000000000004100                                                                                                                                                                                
 x12: 0x0000000000000004                                                                                                                                                                                
 x13: 0x0000000001000004                                                                                                                                                                                
 x14: 0x00000000000027fb                                                                                                                                                                                
 x15: 0x0000000000002af8                                                                                                                                                                                
 x16: 0xffff00019f97d1c0 [rxR,0xffff00019f956000-0xffff00019f97e000] (__stop_set_sysctl_set + 520)                                                                                                      
 x17: 0xffff0000006749e1 [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (vdrop + 0)                                                                                                               
 x18: 0xffff00019eb337d0 [rwxRW,0xffff00019eb30000-0xffff00019eb35000]                                                                                                                                  
 x19: 0xffff00019eb33c50 [rwxRW,0xffff00019eb33c50-0xffff00019eb33c60]                                                                                                                                  
 x20: 0x0000000000000000                                                                                                                                                                                
 x21: 0xffff00019eb33c60 [rwxRW,0xffff00019eb33c50-0xffff00019eb33c60]                                                                                                                                  
 x22: 0x0000000000000000                                                                                                                                                                                
 x23: 0xffff00019f97c360 [rwRW,0xffff00019f97c360-0xffff00019f97c880] (null_vnodeops + 0)                                                                                                               
 x24: 0xffff00000101b300 [rwRW,0xffff00000101b300-0xffff00000101b350] (vop_open_desc + 0)                                                                                                               
 x25: 0xffff00019eb33830 [rwxRW,0xffff00019eb33830-0xffff00019eb33930]                                                                                                                                  
 x26: 0xffff00019eb33930 [rwxRW,0xffff00019eb33930-0xffff00019eb33a30]                                                                                                                                  
 x27: 0x0000000000000000                                                                                                                                                                                
 x28: 0x0000000000000010                                                                                                                                                                                
 x29: 0xffff00019eb33a90 [rwxRW,0xffff00019eb30000-0xffff00019eb35000]                                                                                                                                  
 ddc: 0x0000000000000000                                                                                                                                                                                
  sp: 0xffff00019eb337d0 [rwxRW,0xffff00019eb30000-0xffff00019eb35000]                                                                                                                                  
  lr: 0xffff00019f96a69d [rxR,0xffff00019f956000-0xffff00019f97e000] (sentry) (null_open + 28)                                                                                                          
 elr: 0xffff00019f96a21c [rxR,0xffff00019f956000-0xffff00019f97e000] (null_bypass + df)                                                                                                                 
spsr:         64400045                                                                                                                                                                                  
 far: ffff00019eb33c60                                                                                                                                                                                  
 esr:         9600002a                                                                                                                                                                                  
WARNING !list_empty(&lock->head) failed at /usr/src/sys/dev/drm/core/drm_modeset_lock.c:268                                                                                                             
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /usr/src/sys/dev/drm/core/drm_atomic_helper.c:617                                                                                                
WARNING !drm_modeset_is_locked(&dev->mode_config.connection_mutex) failed at /usr/src/sys/dev/drm/core/drm_atomic_helper.c:667                                                                          
[... cut ...]
--- exception, esr 0x56000000                                                                                                                                                                           
KDB: enter: panic                                                                                                                                                                                       
[ thread pid 20804 tid 100149 ]                                                                                                                                                                         
```

jrtc27, 2022-11-09:
The printed panic isn’t really the root cause. The problem is the cap_abort that called panic; what you see is the nested panic because the DRM code is not very robust and doesn’t handle panics properly.

kwitaszczyk, 2022-11-09:
@jrtc27 Thanks. I disabled DRM, tested it again and I can confirm it's an issue related to nullfs.

closed 2022-11-11 by kwitaszczyk