CTSRD-CHERI/cheribsd #1553: drm_gem mutex warning

rwatson, 2022-11-18:
With WITNESS and INVARIANTS enabled, and running with a pure-capability kernel from the 11 November 2022 CheriBSD development snapshot, I intermittently see the following warning:

```WARNING !mutex_is_locked(&obj->dev->struct_mutex) failed at /local/scratch/jenkins/workspace/CheriBSD-pipeline_dev@3/cheribsd/sys/dev/drm/core/drm_gem.c:1048```

I’m not certain what is triggering this warnings. Wayland is running on the video console, but I’m not in my office so it’s not a specific thing driven by user activity.

rwatson, 2022-11-19:
Tagging @gvnn3.

bukinr, 2022-11-30:
https://github.com/CTSRD-CHERI/cheribsd/pull/1579

brooksdavis, 2022-12-17:
https://github.com/CTSRD-CHERI/cheribsd/pull/1579 was merged and fixed this

closed 2022-12-17 by brooksdavis