CTSRD-CHERI/cheribsd #1564: Nested panic due to DRM

jrtc27, 2022-11-28:
The following diff was proposed by Ruslan but has not been put up as a PR:
```
diff --git a/sys/dev/drm/core/drm_atomic_helper.c b/sys/dev/drm/core/drm_atomic_helper.c
index 85d163f1680..e45f8d0ab9b 100644
--- a/sys/dev/drm/core/drm_atomic_helper.c
+++ b/sys/dev/drm/core/drm_atomic_helper.c
@@ -1598,7 +1598,8 @@ void drm_atomic_helper_commit_tail_rpm(struct drm_atomic_state *old_state)
 
        drm_atomic_helper_commit_hw_done(old_state);
 
-       drm_atomic_helper_wait_for_vblanks(dev, old_state);
+       if (!SKIP_SLEEP())
+               drm_atomic_helper_wait_for_vblanks(dev, old_state);
 
        drm_atomic_helper_cleanup_planes(dev, old_state);
 }
diff --git a/sys/dev/drm/core/drm_fb_helper.c b/sys/dev/drm/core/drm_fb_helper.c
index 3844340d9ff..8dc9b87f31d 100644
--- a/sys/dev/drm/core/drm_fb_helper.c
+++ b/sys/dev/drm/core/drm_fb_helper.c
@@ -510,7 +510,8 @@ static int restore_fbdev_mode_atomic(struct drm_fb_helper *fb_helper, bool activ
 
        drm_atomic_state_put(state);
 out_ctx:
-       drm_modeset_drop_locks(&ctx);
+       if (!SKIP_SLEEP())
+               drm_modeset_drop_locks(&ctx);
        drm_modeset_acquire_fini(&ctx);
 
        return ret;
```

bukinr, 2022-11-30:
https://github.com/CTSRD-CHERI/cheribsd/pull/1569

gvnn3, 2022-12-14:
Closed by #1585

commit 142e0ce4 (2022-12-13): Work around nested panic in DRM

When a kernel panic occurs while DRM graphics are enabled a second panic
occurs before the message can be printed or a backtrace produced.  This
prevents the users from seeing the panic message.

This change works around the panic in two parts:
 - dont wait for vblank during panic.
 - avoid call to drm_modeset_drop_locks as well due to unknown bug

This should really be done in a FreeBSD-specific layer.

Fixes #1564
commit 66c83b5d (2022-12-13): Work around nested panic in DRM

When a kernel panic occurs while DRM graphics are enabled a second panic
occurs before the message can be printed or a backtrace produced.  This
prevents the users from seeing the panic message.

This change works around the panic in two parts:
 - dont wait for vblank during panic.
 - avoid call to drm_modeset_drop_locks as well due to unknown bug

This should really be done in a FreeBSD-specific layer.

Fixes #1564
PR #1585 (2022-12-13): Work around nested panic in DRM
When a kernel panic occurs while DRM graphics are enabled a second panic occurs before the message can be printed or a backtrace produced.  This prevents the users from seeing the panic message.

This change works around the panic in two parts:
 - dont wait for vblank during panic.
 - avoid call to drm_modeset_drop_locks as well due to unknown bug

This should really be done in a FreeBSD-specific layer.

Fixes #1564
commit 559d552f (2022-12-13): Work around nested panic in DRM

When a kernel panic occurs while DRM graphics are enabled a second panic
occurs before the message can be printed or a backtrace produced.  This
prevents the users from seeing the panic message.

This change works around the panic in two parts:
 - dont wait for vblank during panic.
 - avoid call to drm_modeset_drop_locks as well due to unknown bug

This should really be done in a FreeBSD-specific layer.

Fixes #1564
closed 2022-12-14 by gvnn3