CTSRD-CHERI/cheribsd #1572: backtrace_fmt() returns array of non-tagged pointers

bsdphk, 2022-11-29:
On morello18 (a29002fdf4a), ``backtrace_symbols_fmt()`` returns an array of non-tagged pointers.

Output:

    backtrace = 1, [0] = 0x110c69 [rxR,0x100000-0x1311c0] (sentry)
    backtrace_symbols_fmt() = 0x40935000 [rwRW,0x40935000-0x40935050], [0] = 0x40935010
    In-address space security exception (core dumped)

Program:

    #include <sys/cdefs.h>
    #include <execinfo.h>
    #include <stdio.h>
    #include <string.h>

    #define LEVELS 8

    void
    something(void)
    {
        void *array[LEVELS];
        char **fmt;
        int i;

        memset(array, 0, sizeof array);
        i = backtrace(array, LEVELS);
        printf("backtrace = %d, [0] = %#p\n", i, array[0]);
        fmt = backtrace_symbols_fmt(array, i, "%a %n %d %D %f");
        printf("backtrace_symbols_fmt() = %#p, [0] = %#p\n", fmt, fmt[0]);
        printf("*[0] = %s\n", fmt[0]);
    }

    int
    main()
    {
        setbuf(stdout, NULL);
        something();
        return(0);
    }

arichardson, 2022-11-29:
Not looked through the code in detail, but the comment `Change offsets to pointers` at https://github.com/CTSRD-CHERI/cheribsd/blob/4511a9b9d97ab3ca8041873e353d818d1541d8fc/contrib/libexecinfo/backtrace.c#L222 almost certainly indicates CHERI-incompatible code.

jrtc27, 2022-11-29:
Should be
```C
((char **)(void *)ptr)[j] = ptr + (ptrdiff_t)(intptr_t)((char **)(void *)ptr)[j];
```
as it's all offsets relative to ptr.