CTSRD-CHERI/cheribsd #1577: IOMMU crash when starting SDDM

kwitaszczyk, 2022-11-30:
`service onestart sddm` with CheriABI SDDM installed with packages results in a panic in IOMMU code:
```
panic() at panic+0x40
iommu_pmap_remove_pages() at iommu_pmap_remove_pages+0x340
panfrost_mmu_pgtable_free() at panfrost_mmu_pgtable_free+0x2c
panfrost_postclose() at panfrost_postclose+0x14
drm_file_free() at drm_file_free+0x1ec
drm_release() at drm_release+0xac
drm_fstub_close() at drm_fstub_close+0x74
_fdrop() at _fdrop+0x30
closef() at closef+0x20c
fdescfree() at fdescfree+0x518
exit1() at exit1+0x4bc
sigexit() at sigexit+0x158
postsig() at postsig+0x240
ast() at ast+0x30c
handle_el0_sync() at handle_el0_sync+0x5c
```

kwitaszczyk, 2022-11-30:
@bukinr reproduced this issue and says it's related to:
```
diff --git a/sys/arm64/iommu/iommu_pmap.c b/sys/arm64/iommu/iommu_pmap.c
index eb1de118dd6..a848e87f8c9 100644
--- a/sys/arm64/iommu/iommu_pmap.c
+++ b/sys/arm64/iommu/iommu_pmap.c
@@ -863,7 +863,7 @@ iommu_pmap_remove_pages(pmap_t pmap)
                                        l3e = l3[l];
                                        if ((l3e & ATTR_DESCR_VALID) == 0)
                                                continue;
-                                       panic("%s: l3e found for va %jx\n",
+                                       printf("%s: l3e found for va %jx\n",
                                            __func__, sva);
                                }
```

bukinr, 2022-11-30:
https://github.com/CTSRD-CHERI/cheribsd/pull/1578

brooksdavis, 2022-12-13:
I've merged #1578

closed 2022-12-13 by brooksdavis