CTSRD-CHERI/cheribsd #1584: Buffer overflow in rand()

arichardson, 2022-12-12:
Saw this while trying to run the glib testsuite:
```
Thread 6 received signal SIGPROT, CHERI protection violation.
Capability bounds fault.
[Switching to LWP 126233 of process 27913]
random_r (estate=0x41eae000 [rwRW,0x41eae000-0x41eae0d0]) at /usr/src/lib/libc/stdlib/random.c:471
471			*f += *r;
(gdb) bt
#0  random_r (estate=0x41eae000 [rwRW,0x41eae000-0x41eae0d0]) at /usr/src/lib/libc/stdlib/random.c:471
#1  0x0000000040647794 in rand () at /usr/src/lib/libc/stdlib/rand.c:85
#2  0x000000000011176c in thread_func (arg=0x1fbc900 <tdata+32000256> [rwRW,0x138000-0x3e44000])
    at ../../glib/glib/tests/slice-concurrent.c:60
#3  0x00000000403a2ba0 in g_thread_proxy (data=0x40c04b00 [rwRW,0x40c04800-0x40c04fe0]) at ../../glib/glib/gthread.c:831
#4  0x0000000040498af0 in thread_start (curthread=0x40bf8600 [rwRW,0x40bf8600-0x40bf8fd0])
    at /usr/src/lib/libthr/thread/thr_create.c:319
#5  0x0000000040498600 in _pthread_create (thread=0x40c04b60 [rwRW,0x40c04800-0x40c04fe0], attr=<optimized out>,
    start_routine=<optimized out>, arg=<optimized out>) at /usr/src/lib/libthr/thread/thr_create.c:214
#6  0x000000004049a8dc in _libpthread_init (curthread=0x40184d70 <rtld_bind_lock> [rwRWE,0x40184d70-0x40184d80])
    at /usr/src/lib/libthr/thread/thr_init.c:363
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

```
(gdb) p f
$3 = (uint32_t *) 0x41eae054 [rwRW,0x41eae000-0x41eae0d0]
(gdb) p r
$4 = (uint32_t *) 0x41eae0d0 [rwRW,0x41eae000-0x41eae0d0]
```

jrtc27, 2022-12-12:
This doesn't make sense as a state to be in. It's supposed to be using a TYPE_3 generator, which is meant to have f and r separated by 3*4 bytes at all times within the trailing flexible array minus the first element, which starts at offset 0x50 (so rst_state should point to offset 0x54), but you have the two congruent modulo the degree (31). Given the backtrace involves threads, is glib just being bad and trying to use rand(3) concurrently? It's not a thread-safe interface, and if multiple threads were hammering on rand(3) you could end up with overlapping pointer manipulations that lead to inconsistent state being seen in other threads and then written back. Not to mention the weak memory model making things even worse.

jrtc27, 2022-12-12:
... yes, it is, it starts N_THREADS == 8 threads in parallel, each of which calls rand(3) up to 6 times per iteration of its loop, which executes N_ALLOCS == 50000 times. It's a miracle this test doesn't crash and burn with random corruption on other systems.

arichardson, 2022-12-13:
Not a bug - using rand_r() fixes the issues.

closed 2022-12-13 by arichardson