CTSRD-CHERI/cheribsd #1595: clang error during buildworld

sid-agrawal, 2022-12-17:
I am building `cheribsd` on the `morello-hardware` and running into the following issue during `buildworld`.

Any thoughts on how to proceed?

Commit:
```git
main 8993e1c3bba52a2610f863d206b319904c22e121
```
Error:
```bash
[...]
--- thr_setprio.o ---
/usr/local64/llvm-morello/bin/clang -target aarch64-unknown-freebsd14.0 --sysroot=/usr/obj/usr/home/siagraw/cheribsd/arm64.aarch64c/tmp -B/var/empty -g -O2 -pipe -fno-common -march=morello -Xclang -morello-vararg=new -mabi=purecap   -D__LP64__=1 -DPTHREAD_KERNEL -I/usr/home/siagraw/cheribsd/lib/libc/include -I/usr/home/siagraw/cheribsd/lib/libc/aarch64 -I/usr/home/siagraw/cheribsd/lib/libthr/thread -I/usr/home/siagraw/cheribsd/lib/libthr/arch/aarch64/include -I/usr/home/siagraw/cheribsd/lib/libthr/sys -I/usr/home/siagraw/cheribsd/libexec/rtld-elf -I/usr/home/siagraw/cheribsd/libexec/rtld-elf/aarch64 -I/usr/home/siagraw/cheribsd/lib/libthread_db -fexceptions -D_PTHREAD_FORCED_UNWIND -D_PTHREADS_INVARIANTS -I/usr/home/siagraw/cheribsd/lib/libsimple_printf -DSIMPLE_PRINTF_PREFIX=_thread -g -gz=zlib -MD  -MF.depend.thr_setprio.o -MTthr_setprio.o -std=gnu11 -Wno-format-zero-length -nobuiltininc -idirafter /usr/local64/llvm-morello/lib/clang/13.0.0/include -Werror=implicit-function-declaration -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Wnested-externs -Wold-style-definition -Wno-pointer-sign -Wno-error=pass-failed -Wmissing-variable-declarations -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-error=unused-but-set-variable  -Qunused-arguments    -c /usr/home/siagraw/cheribsd/lib/libthr/thread/thr_setprio.c -o thr_setprio.o
--- cddl/lib/libspl__L ---
clang-13: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-13: note: diagnostic msg: /tmp/atomic-15b9b5.c
clang-13: note: diagnostic msg: /tmp/atomic-15b9b5.sh
clang-13: note: diagnostic msg:

********************
*** [atomic.o] Error code 134

make[4]: stopped in /usr/home/siagraw/cheribsd/cddl/lib/libspl
1 error

make[4]: stopped in /usr/home/siagraw/cheribsd/cddl/lib/libspl

make[3]: stopped in /usr/home/siagraw/cheribsd
--- cddl/lib/libicp_rescue__L ---

make[3]: stopped in /usr/home/siagraw/cheribsd
--- cddl/lib/libicp__L ---

make[3]: stopped in /usr/home/siagraw/cheribsd
--- lib/libthr__L ---

make[3]: stopped in /usr/home/siagraw/cheribsd

make[2]: stopped in /usr/home/siagraw/cheribsd
      198.92 real       611.58 user       154.03 sys

make[1]: stopped in /usr/home/siagraw/cheribsd

make: stopped in /usr/home/siagraw/cheribsd
```

The commands are
```bash
git clone git@github.com:CTSRD-CHERI/cheribsd.git
cd cheribsd

sudo make CROSS_TOOLCHAIN=/usr/local64/share/toolchains/llvm-morello.mk buildworld -j 4 
```
Current Version Installed:
```bash 
uname -a
FreeBSD morello14-dev 14.0-CURRENT FreeBSD 14.0-CURRENT #0 ed75299cb9f: Tue Jul  5 22:33:01 UTC 2022     jenkins@ctsrd-build-linux-xx:/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd-morello-purecap-build/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd/arm64.aarch64c/sys/GENERIC-MORELLO arm64
```

sid-agrawal, 2022-12-17:
[atomic-15b9b5.sh.txt](https://github.com/CTSRD-CHERI/cheribsd/files/10252445/atomic-15b9b5.sh.txt)
[atomic-15b9b5.c.txt](https://github.com/CTSRD-CHERI/cheribsd/files/10252446/atomic-15b9b5.c.txt)

jrtc27, 2022-12-17:
Known bug in that old Morello LLVM. Use a newer version, build an older version of CheriBSD or build with -DWITHOUT_CDDL. We've talked about possibly providing a newer Morello LLVM package for 22.05 so people who know what they're doing can do (unsupported) in-place upgrades but haven't yet as the focus has been on finalising 22.12.

(Note though your error log doesn't include the actual error, just that atomic.o failed to build... but luckily we've seen that before)

closed 2022-12-17 by jrtc27

sid-agrawal, 2022-12-17:
I need the 22.12 version of the kernel, then what is the best way for us to move the machine to the 22.12 kernel?

I am a little confused. There seems to be a catch 22 here, as build world would need the latest llvm and looks like it is also the one which makes it.

Are you suggesting that there is some intermediate version of llvm base which would suffice?

jrtc27, 2022-12-17:
Hm? You don’t get a compiler from buildworld on CheriBSD. Like I said, just force pkg64 to install the latest version from the 22.12 repo and then do your build.

sid-agrawal, 2022-12-17:
The commands on slack seem like they are for 22.05(assuming it is YYYY-MM-DD).

```bash
sudo ABI=CheriBSD:20220511:aarch64 pkg64 update
sudo ABI=CheriBSD:20220511:aarch64 pkg64 upgrade llvm-base
```

I tried different last two digits to try to guess the 22.12 repo, for instance, `CheriBSD:202212XX`. 

I am not sure what is the exact 22.12 repo.

jrtc27, 2022-12-17:
Oops, I copied the wrong one, you want 20220828 (the value of __CheriBSD_version in 22.12's sys/param.h)

sid-agrawal, 2022-12-17:
I was able to update the llvm-base :) But I still got the same build error in `make buildworld`.

However the `make buildword` succeeded when I ran it with `-DWITHOUT_CDDL`. 
A cursory search of CDDL lead me to `man -S5 src.conf`, I don't suppose not having those components will matter.

jrtc27, 2022-12-17:
Maybe upgrading llvm-base didn't upgrade llvm-morello itself? I don't think there's a versioned dependency so probably it just didn't do that; llvm-base is just a wrapper script around llvm that sits in /usr/bin/cc and calls /usr/local64/bin/clang, which itself is a symlink to llvm-morello (/usr/local64/bin/clang-morello).

Not having CDDL will just mean you won't be able to use the experimental ZFS support recently merged.