CTSRD-CHERI/cheribsd #1616: libusb blocks indefinitely

GrassHopper1977, 2023-01-05:
# Summary
libusb appears to block indefinitely when executing `libusb_bulk_transfer()`. The debug messages appear to indicate that this occurs in `libusb10_handle_events_sub`. The problem doesn't seem to occur on Linux or on FreeBSD.

# Background

1. We've been playing with the [inno-maker USB2CAN device](https://github.com/INNO-MAKER/usb2can). It's a generic USB to CAN bus device based on the MCP2518FD device. It has drivers for Windows and Linux but not FreeBSD. The source for the Linux driver can be found [here](https://github.com/torvalds/linux/blob/master/drivers/net/can/usb/gs_usb.c).
2. Originally, we intended to write our own driver for the device as FreeBSD makes writing USB drivers, relatively, simple. Instead we came across [libusb](https://libusb.info/) which is supported by FreeBSD, Linux, Windows and has an implementation for CheriBSD.
3. We used WireShark on Ubuntu Linux to sniff the USB packets and combined this with information derived from the Linux driver to write a simple test application on Linux.
4. We then moved the Test Application to CheriBSD on the Morello board to try it out and it didn't work.
5. We tried it on V22.05 PureCaps build running both the Hybrid and PureCaps kernels and both reacted in the same way.
6. The Hybrid Kernel:
```
FreeBSD cheribsd 14.0-CURRENT FreeBSD 14.0-CURRENT #0 ed75299cb9f: Tue Jul  5 22:33:01 UTC 2022     jenkins@ctsrd-build-linux-xx:/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd-morello-purecap-build/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd/arm64.aarch64c/sys/GENERIC-MORELLO arm64
```
7. The Pure Caps Kernel:
```
FreeBSD cheribsd 14.0-CURRENT FreeBSD 14.0-CURRENT #0 ed75299cb9f: Tue Jul  5 22:34:51 UTC 2022     jenkins@ctsrd-build-linux-xx:/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd-morello-purecap-build/local/scratch/jenkins/workspace/CheriBSD-pipeline_releng_22.05/cheribsd/arm64.aarch64c/sys/GENERIC-MORELLO-PURECAP arm64
```
8. We then set up FreeBSD V12.3 on an old laptop that we had knocking around and it worked as expected.
9. The FreeBSD version:

```
FreeBSD test.freebsd.local 12.3-RELEASE FreeBSD 12.3-RELEASE r371126 GENERIC amd64
```

# Test Application
This is just some "quick & dirty" code to prove the concept works. If it had worked we would've spent some time writing a neater implementation.

## Basic Operation
1. Displays the USB devices connected to the system
2. Looks for a device of the correct type
3. Disconnects any existing OS drivers
8. Connects to the device and configures the speed to 500 kbps.
10. Reads 10 CAN messages.
11. Writes a single CAN message to the bus.
12. Close the USB port
13. Reconnects any existing drivers
14. Exits

## The Source
This is the source that was run on both FreeBSD and CheriBSD (the Linux version had some different includes).
```c
#include <stdio.h>
#include <string.h>
#include "libusb.h"

#define USB_VENDOR_ID   0x1D50
#define USB_PRODUCT_ID  0x606F

#define ENDPOINT_FLAG_IN        0x80
#define ENDPOINT_IN     (0x01 | ENDPOINT_FLAG_IN)
#define ENDPOINT_OUT    0x02

int read_packet(struct libusb_device_handle *devh) {
        unsigned char data[24];
        int len = 0;
        int ret = libusb_bulk_transfer(devh, ENDPOINT_IN, data, sizeof(data), &len, 1000);
        if(ret == 0) {
                printf(" IN (%2d):", len);
                for(int i = 0; i < len; i++) {
                        printf(" %02x", data[i]);
                }
                printf("\n");
        } else {
                printf("%s: %s\n", libusb_error_name(ret), libusb_strerror(ret));
        }

        return ret;
}

[... 394 lines cut ...]
Setting speed to 500kps
Opening the port
Reading packets...
 1. LIBUSB_FUNCTION: libusb_bulk_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer leave 0
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
LIBUSB_FUNCTION: libusb10_handle_events_sub enter
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed exit
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
LIBUSB_FUNCTION: libusb10_handle_events_sub enter

```

## Things That We've Tried
1. Changing the timeout (final parameter on the call to `libusb_bulk_transfer()`) to various other settings. Even a timeout of 0 makes no difference.
2. We also wrote a version using the non-blocking calls but it blocked in the same place (`LIBUSB_FUNCTION: libusb10_handle_events_sub enter`

# Conclusion

- It appears to be an issue with CheriBSD but we've run out of time to investigate further at this point. We need to move on with our project so we will be using a different CAN solution. However, this indicates that there may be a problem in `libusb10_handle_events_sub()` so it may effect other users in the future.
- The configuration messages appear to work correctly, it's only when we attempt to read or write data that the issue occurs.

Note: We will be upgrading our Morello to V22.12 tomorrow, so will will retry some of this tests then.
p.s. We used the same physical USB device on each OS that we tried it on.

jrtc27, 2023-01-05:
Is this true for both hybrid and pure-capability userspace (-mabi=aapcs vs -mabi=purecap)?

GrassHopper1977, 2023-01-06:
It appears to work in Hybrid mode.
To be honest, I made so (naive) assumptions:
1. I was under the impression that I couldn't compile for Hybrid when running the purecaps build but, if the the purecaps kernel defaults to Hybrid mode, then of course it must be possible.
2. I didn't think that a Hybrid build would execute when running the purecaps kernel but it, clearly, does.
3. I thought that any issues between hybrid mode and purecaps would cause exceptions but it appears to be more subtle than that.
4. I assumed that the error would be on the OS level but whatever is going wrong is actually happening inside the app that I've built.

### Compilation Instruction

```
cc -v -g -O2 -Wall -mabi=aapcs -cheri-bounds=subobject-safe -lusb  -o testusb testusb.c
```

### Output (Successful)

```
./testusb
LIBUSB_API_VERSION = 01000102
LibUSB version = 1.0.0.2016  https://www.freebsd.org
Trying libusb_init...
6 USB devices found.

USB Devices found:
 1 Vendor ID: 0, Product ID: 0, Manufacturer: 1, Product: 2, Serial: 0
 2 Vendor ID: 1507, Product ID: 1544, Manufacturer: 0, Product: 1, Serial: 0
 3 Vendor ID: 1507, Product ID: 1544, Manufacturer: 0, Product: 1, Serial: 0
 4 Vendor ID: 1027, Product ID: 24577, Manufacturer: 1, Product: 2, Serial: 3
 5 Vendor ID: 1027, Product ID: 24577, Manufacturer: 1, Product: 2, Serial: 3
 6 Vendor ID: 7504, Product ID: 24687, Manufacturer: 1, Product: 2, Serial: 3
Attempting to open the USB2CAN device with 1d50:606f now...
LIBUSB_FUNCTION: libusb_open_device_with_vid_pid enter
LIBUSB_FUNCTION: libusb_open_device_with_vid_pid leave
Device opened.
Checking if kernel driver is active.
Endpoints found: 129 (0x81), 2 (0x02),
Setting up...
Setting up the port 0
Setting up the port 1
Setting up the port 2
Data: 00 00 00 00 02 00 00 00 01 00 00 00
Setting up the port 3
Data: F3 00 00 00 00 6C DC 02 01 00 00 00 10 00 00 00 01 00 00 00 08 00 00 00 04 00 00 00 01 00 00 00 00 04 00 00 01 00 00 00
Setting speed to 500kps
Opening the port
Reading packets...
 1. LIBUSB_FUNCTION: libusb_bulk_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer leave 0
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
LIBUSB_FUNCTION: libusb10_handle_events_sub enter
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed exit
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
LIBUSB_FUNCTION: libusb10_handle_events_sub enter
LIBUSB_TRANSFER: sync I/O done
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed exit
LIBUSB_FUNCTION: libusb_bulk_transfer leave
 IN (20): ff ff ff ff 01 01 00 00 08 00 00 00 60 60 ff f5 0a 4a 00 81
 2. LIBUSB_FUNCTION: libusb_bulk_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer leave 0
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
LIBUSB_FUNCTION: libusb10_handle_events_sub enter
LIBUSB_TRANSFER: sync I/O done
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed exit
LIBUSB_FUNCTION: libusb_bulk_transfer leave
 IN (20): ff ff ff ff 03 02 00 00 08 00 00 00 c2 c2 ff 93 6c e8 04 87
 3. LIBUSB_FUNCTION: libusb_bulk_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer enter
LIBUSB_FUNCTION: libusb_submit_transfer leave 0
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed enter
[... 72 lines cut ...]
LIBUSB_FUNCTION: libusb10_handle_events_sub enter
LIBUSB_TRANSFER: sync I/O done
LIBUSB_FUNCTION: libusb_handle_events_timeout_completed exit
LIBUSB_FUNCTION: libusb_bulk_transfer leave
OUT (20): 00 00 00 00 23 01 00 80 08 00 00 00 01 02 03 04 05 06 07 08
Closing the port
Device closed.
Trying libusb_exit...
```

### Further Experiments
### 1. Removing ```-cheri-bounds=subobject-safe``` from the purecaps compilation instruction.
Perhaps the `subobject-safe` is causing one of the structure members to move in some way?
Command line is now: ``cc -v -g -O2 -Wall -lusb  -o testusb testusb.c
The problem is still present.

### 2. Explicitly setting ```-mabi=purecaps``` in the build.
Command line is now `cc -v -g -O2 -Wall -mabi=purecap -cheri-bounds=subobject-safe -lusb  -o testusb testusb.c`
The problem is still present.

### Thoughts

- I've built both a purecap and hybrid version (I've attached the two files in [testusb_builds.zip](https://github.com/CTSRD-CHERI/cheribsd/files/10359044/testusb_builds.zip)).
- I need to try this on V22.12 as well and see if there are any differences.
- I'm not familiar with this assembly yet but I can have a look since I have the hardware required to test this.

kwitaszczyk, 2023-01-09:
@GrassHopper1977 FYI `cc ... -mabi=aapcs ...` results in executing `clang` with both `-mabi=purecap` and `-mabi=aapcs`. While it works as expected due the latter taking precedence over the former, I guess it's worth being aware of that. `/usr/bin/cc` is a shell script that compiles for the native ABI and would compile for the hybrid ABI if you used CheriBSD with hybrid ABI user space.

jrtc27, 2023-01-09:
> @GrassHopper1977 FYI `cc ... -mabi=aapcs ...` results in executing `clang` with both `-mabi=purecap` and `-mabi=aapcs`. While it works as expected due the latter taking precedence over the former, I guess it's worth being aware of that. `/usr/bin/cc` is a shell script that compiles for the native ABI and would compile for the hybrid ABI if you used CheriBSD with hybrid ABI user space.
> 
> In general, a clean way to compile for an arbitrary ABI is to use `clang -march=morello -mabi=purecap|aapcs -Xclang -morello-vararg=new`.

No, using cc like this is well-defined and to be recommended. Using clang with explicit arguments is worse; this way you don’t have to care about that. And the fact that both -mabi options get passed is an implementation detail that mimics how clang itself should behave (in that the default ABI should be purecap), not something to be concerned about; it is completely intended that it can be overridden like that.