CTSRD-CHERI/cheribsd #1628: Morello flags are not useable today.

jacobbramley, 2023-01-23:
CheriBSD appears to leave `TCR_EL1.TBI0` disabled, which means that existing programs using AArch64 TBI (top byte ignore) cannot be easily ported to CheriBSD on Morello. In addition, it is not easy to detect the presence of this feature dynamically, since it has no `auxv` bit or similar.

I think that all that is required is for the kernel to enable the control bit. It does not need to be tolerant of flagged pointers/capabilities in syscall arguments, and the C library similarly needs no awareness. This level of support is similar to what was provided for Armv8.0 (notably, before MTE). However, there may be other complications that I haven't considered.

jrtc27, 2023-01-23:
> I think that all that is required is for the kernel to enable the control bit.

That's not true, anything looking at FAR_EL1 needs to know about it otherwise you'll get spurious SIGSEGVs.

jrtc27, 2023-01-23:
https://reviews.freebsd.org/D20835 is the old upstream FreeBSD review