CTSRD-CHERI/cheribsd #1637: wpa_supplicant failing to start on Morello

alexdaniellungu, 2023-01-26:
wpa_supplicant is failing to start on Morello, hence no secure Wi-Fi connections can be established. The issue was traced back to an alignment issue in libpcap's newchunk_nolongjmp.

With default libpcap:
ieee80211_load_module: load the wlan_amrr module by hand for now.
wlan0: Ethernet address: 8c:ae:4c:f8:a5:9e
pid 470 (wpa_supplicant), jid 0, uid 0: exited on signal 10 (core dumped)

With updated libpcap, note authmode WPA2:
wlan0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 1500
        ether 8c:ae:4c:f8:a5:9e
        inet 192.168.0.39 netmask 0xffffff00 broadcast 192.168.0.255
        groups: wlan
        ssid "Pretty fly for a Wi-Fi" channel 6 (2437 MHz 11g ht/20) bssid e4:57:40:18:eb:15
        regdomain FCC country US authmode WPA2/802.11i privacy ON
        deftxkey UNDEF AES-CCM 2:128-bit AES-CCM 3:128-bit txpower 30 bmiss 7
        scanvalid 60 protmode CTS ht20 ampdulimit 64k ampdudensity 8 shortgi
        -stbc -ldpc -uapsd wme roaming MANUAL
        parent interface: rtwn0
        media: IEEE 802.11 Wireless Ethernet DS/1Mbps mode 11ng
        status: associated
        nd6 options=29<PERFORMNUD,IFDISABLED,AUTO_LINKLOCAL>

brooksdavis, 2023-01-26:
Thanks for finding this. Could you submit a PR with the fix?

jrtc27, 2023-01-26:
It's just s/sizeof(long)/_Alignof(max_align_t)/g on one line in the function, though I'd like to upstream it to libpcap in a way that works for pre-C99 given that's not a hard requirement (though they do make use of some features that pre-C99 MSVC is happy with).

alexdaniellungu, 2023-01-26:
> Thanks for finding this. Could you submit a PR with the fix?

Please see https://github.com/CTSRD-CHERI/cheribsd/pull/1638 .

kwitaszczyk, 2023-09-12:
I'm closing this issue as it was fixed in dev, main and backported to 22.12.

PR #1665 (2023-03-03): 22.12 backports 20230303
Backport a number of minor fixes (nothing here warrants a point release):
 - Fix __USER_CAP creation from processes in restricted mode (not available by default) and construct capabilities according to the processes DCC offsetting mode.
 - Fix a bug in the strfcap manpage and actually install it
 - Fix builds on macOS Ventura (https://github.com/CTSRD-CHERI/cheribuild/issues/339)
 - Correct an issue with malloc where we returned zero-length capabilities for zero sized requests which could lead to a use-after-free in fairly contrived cases.
 - Correct an alignment issue [... cut]