CTSRD-CHERI/cheribsd #1656: Leftover permissions in relocated capabilities

kevin-brodsky-arm, 2023-02-22:
Looking at `init_cap_from_fragment()` in `libexec/rtld-elf/aarch64/reloc.c`, it seems that the obvious permissions are cleared, but not necessarily all of them. In particular, I would assume that the VMem permission remains, which is undesirable as e.g. pointers to global shouldn't allow `mprotect()`ing the underlying pages.

qwattash, 2023-02-23:
Good point, I suspect we would want to be stricter in the kernel relocation processing as well, although we don't have a meaning for those permission bits within the purecap kernel it is probably better to clear anything that is not explicitly needed.

qwattash, 2023-02-24:
After reviewing this, I think this is not necessary. I'm not sure whether we want to clear SW permission bits for other reasons but the capability given to rtld by the kernel correctly strips the the `SW_VMEM` permission bit so it is impossible for the user process to ever get an `SW_VMEM` capability outside of the `mmap()` interface.
It may be worth to be explicit about other permission bits that may be given a different meaning by the program and we don't know about it. Probably we don't want to randomly use them in global capabilities.

qwattash, 2023-02-27:
Trying to be stricter about this in #1661

kevin-brodsky-arm, 2023-03-20:
Interesting, thanks for these details. I'm not sure I understand why the capabilities that rtld obtains from the kernel don't have the VMem permission though? In the general (DSO) case, they will come from `mmap()`, which surely returns a capability with VMem?

qwattash, 2023-03-20:
Uhm, I think I was mistaken. I was thinking about the capabilities in auxargs, but indeed the capability for relocations for mapped objects comes from `obj->text_rodata_cap` or `obj->relocbase` which were not being cleared. The patch I linked should do the right thing. Probably I should write a test that goes through the whole GOT and checks for permission bits although I'm not sure whether cheribsdtest is the right place.