CTSRD-CHERI/cheribsd #1700: Can not run clang on cheribsd-riscv64-purecap

zintown, 2023-05-20:
I have installed llvm-riscv64-hybrid-for-purecap-rootfs by cheribuild.py, but when I run clang on cheribsd-riscv64-purecap, it failed.
the command is:
```
/opt/riscv64-hybrid/llvm-project/bin/clang-14
```
the error message is:
```
Warning: Attempted to load position-dependent executable with non-representable base: /opt/riscv64-hybrid/llvm-project/bin/clang-14
Abort trap
```
Where is the problem? 
Do I need to add specific compilation options when compiling cheribsd-riscv64-purecap？

jrtc27, 2023-05-20:
@qwattash You added this code in https://github.com/CTSRD-CHERI/cheribsd/pull/913, but it's not ok to enforce this for non-CHERI binaries

qwattash, 2023-07-13:
Closing this as resolved.

PR #1741 (2023-07-13): Do not enforce representability for hybrid binaries in elf imgact.
When loading hybrid position-dependent binaries, we do not care about the representability of the ELF sections. The reservation creation will skip the representability alignment enforcement as well, so avoid complaining about alignment when creating the imgact_capability.

This should fix #1700.
closed 2023-07-13 by qwattash