CTSRD-CHERI/cheribsd #1742: [c18n] Proposal: Embedding function signatures in new ELF section

dpgao, 2023-07-10:
# Background

The library-based compartmentalisation model enables a pair of mutually distrusting function caller and callee to pass data between each other through function arguments and return values while at the same time guarantees that no additional capabilities are leaked at the ABI level.

The current prototype implementation of this model partially fulfils thi guarantee. For example, it assigns different execution stacks to functions in different trust domains, preventing stack corruption attacks and leakage of local variables in deallocated call frames.

The next release of the prototype will provide additional protections. Namely, callers no longer need to rely on the callee for the integrity and confidentiality of the content in callee-saved registers, and both callers and callees can be sure that no capabilities in temporary registers can be leaked through a function call.

One omission, though, is that unused function argument registers and return value registers can still leak capabilities across domains. This proposal aims to address this problem.

# Design

The trampolines that perform domain transitions are made responsible for clearing these registers. However, due to the lack of run-time information about the signature of any particular function, it is currently impossible to identify, at run-time, the unused argument and return value registers for any function.

The compiler should therefore be embed this information, at build time, in the produced ELF binaries. Function callers should at least provide information about which argument registers are unused, and function callees should at least provide information about which return value registers are unused. This makes both sides responsible for their own security. In practice, it is simplest to require callers and callees to provide _complete_ information, and a mismatch at run-time should trigger an error that serves as a good way to detect certain function signature mismatches.

A new ELF section, tentatively named `.c18n.signature`, will be generated by the compiler toolchain. This section contains an array of elements where the `i`-th element corresponds to entry `i` of the dynamic symbol table (`.dynsym`).

Each element has the following structure
```
struct Elf_Signature {
    unsigned int sig_valid : 1;
    unsigned int sig_rargs : 4;
    ungigned int sig_margs : 1;
    unsigned int sig_ret : 2;
    char reserved[3];
};
```

* `sig_valid` bit is set when `sig_args` and `sig_ret` contain meaningful values.
* `sig_rargs` encodes the number of register arguments taken by the function. On Morello, this number ranges from 0 to 8.
* `sig_margs` encodes whether the function takes memory arguments. On Morello, this indicates whether register `c9` is used to pass arguments.
* `sig_ret` encodes how the function passes its return values.
    * `00`: the function does not return any value.
    * `01`: the function returns by writing to a buffer supplied by the caller through `c8`.
    * `10`: the function returns a value through registers `c0` and `c1`.
    * `11`: the function returns a values through register `c0`.

kwitaszczyk, 2023-07-11:
I have a couple of questions to understand better what's expected here.

1.  Is a `Elf_Signature` entry emitted for each non-inlined function or only functions appearing in relocations?

1. > and a mismatch at run-time should trigger an error that serves as a good way to detect certain function signature mismatches

   Do you mean a mismatch between `Elf_Signature` and arguments that are actually passed (similarly, returned values) at run time? If so, how that could be detected at run time?

1. > sig_valid bit is set when `sig_*` contain meaningful values.

   When `sig_*` don't contain meaningful values?

1. > `00`: the function returns values through registers `c0` and `c1`.

   Is that the most common case? I'd make `00` the most common case that would match the behaviour when `Elf_Signature` is not found.

1. > `11`: the function returns by writing to a buffer supplied by the caller through `c8`.

   I'd rephrase that not to specific to Morello in the general case: "the function returns values by writing to memory. On Morello, this indicates `c8` contains a reference to a buffer with a result".

1. > char reserved[3];

   What's the use case for having the reserved bytes? Are you expecting changes that would reuse these bytes while being compatible with ABIs before and after a change?

dpgao, 2023-07-11:
> On 11 Jul 2023, at 11:33, Konrad Witaszczyk ***@***.***> wrote:
> 
> 
> I have a couple of questions to understand better what's expected here.
> 
> 	• Is a Elf_Signature entry emitted for each non-inlined function or only functions appearing in relocations?

It is emitted for each entry in the dynamic symbol table (`.dynsym`), which includes both exported symbols and imported symbols (which also appear relocations).

> • and a mismatch at run-time should trigger an error that serves as a good way to detect certain function signature mismatches
>
> Do you mean a mismatch between Elf_Signature and arguments that are [...]

Sorry for being unclear. I meant if the caller and callee have different Elf_Signatures for the same symbol, then an error should be raised.

> 	• sig_valid bit is set when sig_* contain meaningful values.
> 
> When sig_* don't contain meaningful values?

Not every symbol in `.dynsym` is a function symbol—some of them are object symbols. For entries in `.c18n.signature` that correspond to non-function symbols, the `sig_valid` bit is set to 0 to indicate that the entry is meaningless.

> 	• 00: the function returns values through registers c0 and c1.
> 
> Is that the most common case? I'd make 00 the most common case that would match the behaviour when Elf_Signature is not found.

Probably not, but if Elf_Signature is not found, we can simply default to 0b00000000. And since the sig_valid bit here is 0, the remaining bits will be ignored.

> 	• 11: the function returns by writing to a buffer supplied by the caller through c8.
> 
> I'd rephrase that not to specific to Morello in the general case: "the function returns values by writing to memory. On Morello, this indicates c8 contains a reference to a buffer with a result".

Well, I not sure if this encoding format can be (or should be) portable to other ABIs. We might need a different format for each platform anyway, so being Morello-specific here doesn’t hurt.

> 	• char reserved[3];
> 
> What's the use case for having the reserved bytes? Are you expecting changes that would reuse these bytes while being compatible with ABIs before and after a change?

I don’t have any use cases in mind but it seems a good idea to reserve some space in case we want to associate more information with each dynamic symbol.

kwitaszczyk, 2023-07-11:
> It is emitted for each entry in the dynamic symbol table (`.dynsym`), which includes both exported symbols and imported symbols (which also appear relocations).

Apologies, I just noticed you mentioned `.dynsym` before.

In the kernel, a kernel module registers itself within a subsystem by defining an object that includes function pointers to internal kernel module's routines. Later, the kernel uses that object to branch into the kernel module. AFAIK, the kernel linker is mostly responsible for handling calls from a kernel module into the kernel and between kernel modules but not from the kernel into a kernel module. In such case, the subsystem registering a kernel module is responsible for creating trampolines, not the kernel linker.

From what I can see, `.dynsym` does not include such symbols exported as function pointers within objects (as they're not needed to link a kernel module). They are available in `.symtab` but this section is not always available as it can be stripped. It would be great to come up with a solution that allows to dynamically construct trampolines independently of a linker to internal library/module functions with their `Elf_Signature` entries. 

I imagine this case can also appear in the user space where a program or a library obtains an object from another library with a function pointer to branch into that library. Have you thought about such case? Does anything prevent this in the user space or such calls do not currently use trampolines?

dpgao, 2023-07-11:
> I imagine this case can also appear in the user space where a program or a library obtains an object from another library with a function pointer to branch into that library. Have you thought about [...]

This is a great point. I guess in order to make arbitrary function pointers work we'll have to make an `Elf_Signature` entry for every item in `.symtab`, not just `.dynsym`. This also means that libraries cannot strip this section if they want to be used with `c18n`.

brooksdavis, 2023-07-11:
> I don’t have any use cases in mind but it seems a good idea to reserve some space in case we want to associate more information with each dynamic symbol.

I might be convinced by `char reserved[1]`, but even then I'm skeptical without a clear use case since it's always going to be possible to add `.c18n.signature2` for compatible expansion should we not be able to have another flag day.

dpgao, 2023-07-12:
> I might be convinced by `char reserved[1]`, but even then I'm skeptical without a clear use case since it's always going to be possible to add `.c18n.signature2` for compatible expansion should we not be able to have another flag day.

I'm happy to go with `char reserved[1]`, if only because some ABIs might need more bits to encode register clearing information. But I agree that we can always add more sections if needed.

@brooksdavis Do you have any thoughts on how to deal with indirect calls?

brooksdavis, 2023-07-12:
Could you provide a trivial example of the safebox/sandbox API in expected use? I'm pretty sure that ultimate we don't want to expose them to the user and even if we need to expose them they should be reasonably type preserving.

dpgao, 2023-07-12:
> Could you provide a trivial example of the safebox/sandbox API in expected use? I'm pretty sure that ultimate we don't want to expose them to the user and even if we need to expose them they should be reasonably type preserving.

@brooksdavis Consider the following example where the main executable registers a callback with `libfoo`. The return type of `_rtld_safebox_func` and `_rtld_sandbox_func` is `void *`, so no type casting is needed. We probably need to expose this API to users, but ideally the compiler can insert them automatically.

```
// main.c
static int handler(char *buf, size_t len) {
    // process buffer
    return 0;
}

int main() {
    // before: libfoo_register_handler(handler);
    libfoo_register_handler(_rtld_safebox_func(handler, (struct Elf_Signature) {
        .sig_valid = 1, .sig_rargs = 2, .sig_margs = 0, .sig_ret = 1
    }));
}

// libfoo.c
static int (*handler)(char *, sizet_t);
void libfoo_register_handler(int (*f)(char *, size_t)) {
    // before: handler = f;
    handler = _rtld_sandbox_func(f, (struct Elf_Signature) {
        .sig_valid = 1, .sig_rargs = 2, .sig_margs = 0, .sig_ret = 1
    });
}
```

kwitaszczyk, 2023-07-19:
@dpgao I'd like to discuss some modifications to limit the number of required changes in libraries/modules to make use of function pointer calls in the model, and possibly allow to use the model with closed-source libraries and kernel modules. What do you think about these two changes to this design:

1. Change `_rtld_safebox_func()` to only allow creating a function pointer to a function that is listed in symbol tables visible to a run-time linker, i.e. when compiling a program/kernel or a library/module exportable (from the program/kernel) or importable (from the library/module) functions must be placed in the symbol tables of the program/kernel or the library/module. In such case, a run-time linker could retrieve an `Elf_Signature` entry itself rather than requiring to specify it by the caller of `_rtld_safebox_func()`.

2. Remove `_rtld_sandbox_func()` and allow to directly call a function using a function pointer returned by `_rtld_safebox_func()`. This would disallow the caller of the returned function pointer to indicate what's the `Elf_Signature` entry from their perspective. To discuss that problem, I'd consider two cases:

  * A library/module can be recompiled to use the model. The caller of the returned function pointer must cast it to a function data type that already includes information on the number of function arguments. I wonder if we could change the ABI to emit at compile-time an `Elf_Signature` entry for function calls using function pointers. That `Elf_Signature` entry could be used by a trampoline at run-time.

  * A library/module cannot be recompiled to use the model. The caller of the returned function pointer must trust the function pointer and calls the function using the `Elf_Signature` entry of the owner of the function.