CTSRD-CHERI/cheribsd #1818: mprotect(2) never adds VM_PROT_*_CAP

jrtc27, 2023-08-25:
VM_PROT_ADD_CAP is never called on prot and max_prot. There's a weird VM_MAP_PROTECT_KEEP_CAP dance in vm_map_protect to implicitly add it to prot, but not max_prot, and if you're setting PROT_MAX then it uses the current protection rather than max_protection to determine whether to do VM_PROT_ADD_CAP, so you have to overmap rather than mprotect to get the page table permissions.

```
#include <sys/mman.h>

#include <err.h>
#include <stdio.h>

#define	PAGE_SIZE	4096

int
main(int argc, char **argv)
{
	void * volatile *p;

	p = mmap(NULL, PAGE_SIZE, PROT_NONE | PROT_MAX(PROT_READ | PROT_WRITE),
	    MAP_ANON | MAP_PRIVATE, -1, 0);
	printf("mmap: %#p\n", p);
	if (p == MAP_FAILED)
		return (1);
	if (mprotect((void *)p, PAGE_SIZE, PROT_READ | PROT_WRITE) != 0)
		err(1, "mprotect");
	*p = (void *)p;
	printf("loaded: %#p\n", *p);

	return (0);
}
```

Expected to work but takes a tag-store page fault.

jrtc27, 2023-08-25:
Also messing with those variables in vm_map_protect seems dodgy if you hit the goto again case.

nwf, 2023-08-25:
This seems related to
https://github.com/microsoft/snmalloc/blob/7b597335aedc2c60b0174f6e67a7f7560f841760/src/snmalloc/ds_core/mitigations.h#L243-L246
which I could have sworn had an issue associated with it but if so the association has been lost to time. :(

commit 3dbfc36d (2024-07-30): vm_map_protect: imply VM_PROT_CAP from max_protection

Always check entry->max_protection for VM_PROT_CAP when deciding if we
should imply VM_PROT_CAP for readable and writable pages.

Previously we would only check it if we were setting a new
max_protection (which mprotect generally doesn't do) which meant we
didn't add VM_PROT_CAP when the current permissions were VM_PROT_NONE.

Fixes: #1818
commit ed1c1c8d (2024-07-31): vm_map_protect: imply VM_PROT_CAP from max_protection

Always check entry->max_protection for VM_PROT_CAP when deciding if we
should imply VM_PROT_CAP for readable and writable pages.

Previously we would only check it if we were setting a new
max_protection (which mprotect generally doesn't do) which meant we
didn't add VM_PROT_CAP when the current permissions were VM_PROT_NONE.

Add an assertion that VM_MAP_PROTECT_KEEP_CAP is not set unless we're
actually setting prot or maxprot.

Fixes: #1818
PR #2162 (2024-08-06): vm_map_protect: imply VM_PROT_CAP from max_protection
Always check entry->max_protection for VM_PROT_CAP when deciding if we should imply VM_PROT_CAP for readable and writable pages.

Previously we would only check it if we were setting a new max_protection (which mprotect generally didn't do) which meant we didn't add VM_PROT_CAP when the current permission were VM_PROT_NONE.

Fixes: #1818
commit 29d592e4 (2024-08-06): vm_map_protect: imply VM_PROT_CAP from max_protection

Always check entry->max_protection for VM_PROT_CAP when deciding if we
should imply VM_PROT_CAP for readable and writable pages.

Previously we would only check it if we were setting a new
max_protection (which mprotect generally doesn't do) which meant we
didn't add VM_PROT_CAP when the current permissions were VM_PROT_NONE.

Add an assertion that VM_MAP_PROTECT_KEEP_CAP is not set unless we're
actually setting prot or maxprot.

Fixes: #1818