CTSRD-CHERI/cheribsd #1837: snmalloc on amd64 depends on CMPXCHG16B

brooksdavis, 2023-09-05:
snmalloc currently requires the CMPXCHG16B instruction on x86 which more or less means x86-64-v2. It's not clear that would be acceptable to upstream FreeBSD.

mjp41, 2023-09-14:
@brooksdavis the data structure that makes the requirement is not particularly important in the current design, and could be replaced with a lock without harming perf (this wasn't true in older versions of snmalloc).

brooksdavis, 2023-09-14:
Replacing it would be simplifying.

It also might be fine to increase the FreeBSD standard to requiring CMPXCHG16B or x86-64-v2, but it would need to be approached with care to avoid accidentally and prematurely  killing off large deployments we're unaware of. IIRC David said that both modern Linux and Windows 8 require CMPXCHG16B so I suspect the odds are that we could get away with a change.