CTSRD-CHERI/cheribsd #1848: libunwind tries to unwind beyond pthread_create

arichardson, 2023-09-08:
While testing libunwind/test/thread_unwind.pass.cpp in LLVM I noticed that we get a bounds violation when unwinding beyond the thread start. I will work around the bounds error in libunwind, but this should also be fixed in libthr/thr_create so that libunwind does not attempt to resume unwinding.

Debug output from unwind:
```
found_text && found_unwind in /buildroot/libunwind-riscv64-purecap-build/lib/test-thread_unwind-static-O0
Trying to bound PT_LOAD of .eh_frame in /buildroot/libunwind-riscv64-purecap-build/lib/test-thread_unwind-static-O0
parseInstructions(instructions=0x105339 [rR,0x105328-0x105340]-0x105340, pcoffset=0xffffffffffffffff)
DW_CFA_def_cfa(reg=2, offset=0)
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
parseInstructions(instructions=0x1054b9 [rR,0x1054a8-0x1054c8]-0x1054c8, pcoffset=0x88)
DW_CFA_advance_loc: new offset=4
DW_CFA_def_cfa_offset(1888)
DW_CFA_advance_loc: new offset=12
DW_CFA_offset(reg=1, offset=-16)
DW_CFA_offset(reg=8, offset=-32)
DW_CFA_advance_loc: new offset=16
DW_CFA_def_cfa(reg=8, offset=0)
DW_CFA_nop
DW_CFA_nop
libunwind: __unw_step(cursor=0x417ff830)
getIP: IP = 0x10d8e6 [rxR,0x100000-0x11eb80] (sentry,capmode)
parseInstructions(instructions=0x105339 [rR,0x105328-0x105340]-0x105340, pcoffset=0xffffffffffffffff)
DW_CFA_def_cfa(reg=2, offset=0)
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
parseInstructions(instructions=0x1054b9 [rR,0x1054a8-0x1054c8]-0x1054c8, pcoffset=0x88)
DW_CFA_advance_loc: new offset=4
DW_CFA_def_cfa_offset(1888)
DW_CFA_advance_loc: new offset=12
DW_CFA_offset(reg=1, offset=-16)
DW_CFA_offset(reg=8, offset=-32)
DW_CFA_advance_loc: new offset=16
DW_CFA_def_cfa(reg=8, offset=0)
DW_CFA_nop
DW_CFA_nop
SETTING RETURN REGISTER 1 (ra): 0x10c658 [rxR,0x100000-0x11eb80] (sentry,capmode) 
SETTING CAPABILITY REGISTER 8 (s0): 0x417fff80 [rwRW,0x41400000-0x41800000] 
getIP: IP = 0x10c658 [rxR,0x100000-0x11eb80] (sentry,capmode)
setInfoBasedOnIPRegister(1): pc=0x10c658 [rxR,0x100000-0x11eb80] (sentry,capmode)
Calling dl_iterate_phdr()
parseInstructions(instructions=0x105339 [rR,0x105328-0x105340]-0x105340, pcoffset=0xffffffffffffffff)
DW_CFA_def_cfa(reg=2, offset=0)
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
parseInstructions(instructions=0x105391 [rR,0x105380-0x1053a0]-0x1053a0, pcoffset=0x57)
DW_CFA_advance_loc: new offset=2
DW_CFA_def_cfa_offset(48)
DW_CFA_advance_loc: new offset=6
DW_CFA_offset(reg=1, offset=-16)
DW_CFA_offset(reg=8, offset=-32)
DW_CFA_advance_loc: new offset=8
DW_CFA_def_cfa(reg=8, offset=0)
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
libunwind: step: result = 1
libunwind: __unw_get_proc_info(cursor=0x417ff830, &info=0x417ff6c0)
libunwind: __unw_get_proc_name(cursor=0x417ff830, &buf=0x417ff4c0, bufLen=512)
getIP: IP = 0x10c658 [rxR,0x100000-0x11eb80] (sentry,capmode)
findFunctionName(0x10c658: 0x10c658 [rxR,0x100000-0x11eb80] (sentry,capmode)))
libunwind: unwind_phase2_forced(ex_ojb=0x11db40): start_ip=0x10c600, func=_Z24call_Unwind_ForcedUnwindv, lsda=0x0, personality=0x0
libunwind: __unw_get_reg(cursor=0x417ff830, regNum=-2, &value=0x417ff240)
getReg: -2 = 0x417fff50 [rwRW,0x41400000-0x41800000]
[... 197 lines cut ...]
DW_CFA_nop
DW_CFA_nop
parseInstructions(instructions=0x40156d39 [rR,0x40156d28-0x40156d58]-0x40156d58, pcoffset=0x37e)
DW_CFA_advance_loc: new offset=2
DW_CFA_def_cfa_offset(416)
DW_CFA_advance_loc: new offset=22
DW_CFA_offset(reg=1, offset=-16)
DW_CFA_offset(reg=8, offset=-32)
DW_CFA_offset(reg=9, offset=-48)
DW_CFA_offset(reg=18, offset=-64)
DW_CFA_offset(reg=19, offset=-80)
DW_CFA_offset(reg=20, offset=-96)
DW_CFA_offset(reg=21, offset=-112)
DW_CFA_offset(reg=22, offset=-128)
DW_CFA_offset(reg=23, offset=-144)
DW_CFA_offset(reg=24, offset=-160)
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop
DW_CFA_nop

Thread 2 received signal SIGPROT, CHERI protection violation.
```

arichardson, 2023-09-18:
Maybe all that's needed here is a `.cfi_undefined <return_reg>` in `thread_start`?

bsdjhb, 2023-09-22:
Yes, likely.  However, see my comment on the other PR where it would be really nice if LLVM had a less hacky way of implementing this (e.g. via a new function attribute)

arichardson, 2023-09-22:
I just discovered the [nouwtable](https://clang.llvm.org/docs/AttributeReference.html#nouwtable) attribute that might be sufficient to tell libunwind to stop unwinding (due to missing info).

arichardson, 2024-08-23:
https://reviews.freebsd.org/D46400