CTSRD-CHERI/cheribsd #1924: procstat cheri “Operating not permitted” warnings rather noisy

rwatson, 2023-11-15:
Particularly visible now that we have `-a` support for `procstat cheri`, the `cheri` mode’s warnings about `EPERM` are getting quite distracting:

```
robert@cheri-blossom:~ % procstat -a cheri
  PID COMM                C QUAR  RSTATE
    0 kernel              ?
procstat: sysctl: kern.proc.quarantining: 1: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1: Operation not permitted
    1 init                P    -       -
    2 clock               ?
    3 crypto              ?
    4 cam                 ?
    5 busdma              ?
    6 rand_harvestq       ?
    7 pagedaemon          ?
    8 vmdaemon            ?
    9 bufdaemon           ?
   10 audit               ?
   11 idle                ?
   12 intr                ?
   13 geom                ?
   14 sequencer 00        ?
   15 usb                 ?
   16 vnlru               ?
   17 syncer              ?
procstat: sysctl: kern.proc.quarantining: 626: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 626: Operation not permitted
  626 dhclient            P    -       -
procstat: sysctl: kern.proc.quarantining: 629: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 629: Operation not permitted
  629 dhclient            P    -       -
procstat: sysctl: kern.proc.quarantining: 701: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 701: Operation not permitted
  701 dhclient            P    -       -
procstat: sysctl: kern.proc.quarantining: 902: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 902: Operation not permitted
  902 devd                P    -       -
procstat: sysctl: kern.proc.quarantining: 1123: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1123: Operation not permitted
 1123 syslogd             P    -       -
procstat: sysctl: kern.proc.quarantining: 1168: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1168: Operation not permitted
 1168 daemon              P    -       -
procstat: sysctl: kern.proc.quarantining: 1169: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1169: Operation not permitted
 1169 seatd               P    -       -
procstat: sysctl: kern.proc.quarantining: 1218: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1218: Operation not permitted
 1218 sshd                P    -       -
procstat: sysctl: kern.proc.quarantining: 1222: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1222: Operation not permitted
 1222 cron                P    -       -
procstat: sysctl: kern.proc.quarantining: 1243: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1243: Operation not permitted
 1243 sddm                P    -       -
procstat: sysctl: kern.proc.quarantining: 1248: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1248: Operation not permitted
 1248 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1249: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1249: Operation not permitted
 1249 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1250: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1250: Operation not permitted
 1250 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1251: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1251: Operation not permitted
 1251 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1252: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1252: Operation not permitted
 1252 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1253: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1253: Operation not permitted
 1253 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1254: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1254: Operation not permitted
 1254 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1255: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1255: Operation not permitted
 1255 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1256: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1256: Operation not permitted
 1256 getty               P    -       -
procstat: sysctl: kern.proc.quarantining: 1258: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1258: Operation not permitted
 1258 Xorg                P    -       -
procstat: sysctl: kern.proc.quarantining: 1263: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1263: Operation not permitted
 1263 sddm-helper         P    -       -
procstat: sysctl: kern.proc.quarantining: 1264: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1264: Operation not permitted
 1264 sddm-greeter        P    -       -
procstat: sysctl: kern.proc.quarantining: 1268: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1268: Operation not permitted
 1268 dbus-launch         P    -       -
procstat: sysctl: kern.proc.quarantining: 1269: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 1269: Operation not permitted
 1269 dbus-daemon         P    -       -
procstat: sysctl: kern.proc.quarantining: 3494: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 3494: Operation not permitted
 3494 sshd                P    -       -
procstat: sysctl: kern.proc.quarantining: 3497: Operation not permitted
procstat: sysctl: kern.proc.revoker_state: 3497: Operation not permitted
 3497 sshd                P    -       -
 3498 tcsh                P  yes    none
 3512 procstat            P  yes    none
```

I wonder if we should silence that and print `-` or `???` or something where we can’t present data due to permission denied or similar?

rwatson, 2023-11-15:
I notice that other `procstat` modes prefer `-` when they can’t query values for other users -- e.g., `-c` querying command lines.

brooksdavis, 2023-11-15:
We could just drop the warnings. I'd copied them from other libprocstat code, but they are awkward.

In some ways I think the issue is that `-a` shouldn't show kernel processes at all for `cheri`.

rwatson, 2023-11-15:
Yeah, just not warning seems fine to me.

Showing kernel processes uselessly for `cheri` is consistent with what over procstat modes do? :-)

commit 6bb5b292 (2023-11-16): libprocstat: supress warning on EPERM for caprevoke

The revocation related syscalls return EPERM on processes not owned by
the user are quite spammy with `procstat -a cheri` called by a non-root
user.

Fixes: #1924
closed 2023-11-30 by brooksdavis