CTSRD-CHERI/cheribsd #1932: panic: vm_cheri_revoke post past entry end (with x11/sddm)

kwitaszczyk, 2023-11-24:
With the [2023-11-23 dev snapshot](https://download.cheribsd.org/snapshots/dev/arm64/aarch64c/2023-11-23/]), I get the following panic at the stage I press enter to confirm my password in SDDM:
```
panic: vm_cheri_revoke post past entry end: 50197000 > 50251000 (was 50250000)
cpuid = 0
time = 1700826346
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_trace_self_wrapper() at db_trace_self_wrapper+0x30
vpanic() at vpanic+0x19c
panic() at panic+0x60
vm_cheri_revoke_pass() at vm_cheri_revoke_pass+0x9dc
sys_cheri_revoke() at sys_cheri_revoke+0x39c
do_el0_sync() at do_el0_sync+0x5b8
handle_el0_sync() at handle_el0_sync+0x38
--- exception, esr 0x56000000
```

markjdb, 2023-11-24:
I believe this is fixed by https://github.com/CTSRD-CHERI/cheribsd/pull/1933

markjdb, 2023-11-25:
> I believe this is fixed by #1933

Merged into `dev` as of commit ca0ff3cad39476985e6cb4ff9729ceb1b5a93b21.

closed 2023-11-25 by markjdb