CTSRD-CHERI/cheribsd #1964: `mallocx(..., MALLOCX_ALIGN(a))` returns unaligned buffers for large values of `a`.

jacobbramley, 2023-12-12:
Using the following test function:

```
  void check_mallocx(size_t size) {
    void * data = mallocx(size, MALLOCX_ALIGN(size));
    printf("mallocx(%zx, MALLOCX_ALIGN(%zx)) -> %#lp\n", size, size, data);
    if (!cheri_is_aligned(data, size)) {
      printf(" - Not correctly aligned! offset: %zx\n", cheri_address_get(data) - cheri_address_get(cheri_align_down(data, size)));
    }
  }
```

Passing in a few powers of two:

```
mallocx(400, MALLOCX_ALIGN(400)) -> 0x4060f000 [rwRW,0x4060f000-0x4060f400]
mallocx(800, MALLOCX_ALIGN(800)) -> 0x4061c000 [rwRW,0x4061c000-0x4061c800]
mallocx(1000, MALLOCX_ALIGN(1000)) -> 0x40613000 [rwRW,0x40613000-0x40614000]
mallocx(2000, MALLOCX_ALIGN(2000)) -> 0x40621000 [rwRW,0x40621000-0x40623000]
 - Not correctly aligned! offset: 1000
mallocx(4000, MALLOCX_ALIGN(4000)) -> 0x40635000 [rwRW,0x40635000-0x40639000]
 - Not correctly aligned! offset: 1000
mallocx(8000, MALLOCX_ALIGN(8000)) -> 0x40639000 [rwRW,0x40639000-0x40641000]
 - Not correctly aligned! offset: 1000
mallocx(10000, MALLOCX_ALIGN(10000)) -> 0x40641000 [rwRW,0x40641000-0x40651000]
 - Not correctly aligned! offset: 1000
```

I've seen offsets of `0x2000` in JavaScriptCore, which relies on being able to allocate 16KB-aligned blocks. This causes JSC to fail early on start-up.

It looks like there have been several changes to `mallocx` recently (@brooksdavis?), but I've not bisected to find the culprit.

brooksdavis, 2023-12-12:
The current default mallocx implementation doesn't honor alignment (I missed that it should). Any reason not to use posix_memalign?

brooksdavis, 2023-12-12:
You could also try disabling revocation which will bypass the incorrect implementation. See https://man.cheribsd.org/cgi-bin/man.cgi/mrs#ENABLING_REVOCATION

jacobbramley, 2023-12-12:
Disabling revocation seems to work around it, thanks!

In addition to alignment, JSC is also using `MALLOCX_TCACHE_NONE` and `MALLOCX_ARENA(...)`, and they don't seem to have `posix_memalign` equivalents.

brooksdavis, 2023-12-12:
With mrs we're just ignoring the TCACHE and ARENA flags. I'll add support for alignment. We could probably pass them through, but I'm not convinced they would be particularly meaningful given the effects to having a single, arbitrarily selected thread insert everything into the free list on easy cycle. There's defiantly more work to do here on micro optimization in a quarantined environment.

commit e2c96535 (2023-12-12): mrs: handle MALLOCX_ALIGN in mallocx/rallocx

It turns out that there are consumers of mallocx that use the
alignment bits.  I was cribing from snmalloc's jemalloc compat shims and
those lacked alignment support.

Improve handling of zeroing in mallocx by only zeroing on requestion,
but remain inefficent in rallocx for now.

Fixes: #1964
commit 778d0b4e (2023-12-12): mrs: handle MALLOCX_ALIGN in mallocx/rallocx

It turns out that there are consumers of mallocx that use the
alignment bits.  I was cribing from snmalloc's jemalloc compat shims and
those lacked alignment support.

Improve handling of zeroing in mallocx by only zeroing on requestion,
but remain inefficent in rallocx for now.

Fixes: #1964
commit fbdc24d8 (2023-12-12): mrs: handle MALLOCX_ALIGN in mallocx/rallocx

It turns out that there are consumers of mallocx that use the
alignment bits.  I was cribing from snmalloc's jemalloc compat shims and
those lacked alignment support.

Improve handling of zeroing in mallocx by only zeroing on requestion,
but remain inefficent in rallocx for now.

Fixes: #1964
commit 3f615523 (2023-12-12): mrs: handle MALLOCX_ALIGN in mallocx/rallocx

It turns out that there are consumers of mallocx that use the
alignment bits.  I was cribing from snmalloc's jemalloc compat shims and
those lacked alignment support.

Improve handling of zeroing in mallocx by only zeroing on requestion,
but remain inefficent in rallocx for now.

Fixes: #1964
closed 2023-12-13 by brooksdavis