CTSRD-CHERI/cheribsd #1969: `MALLOCX_ARENA` is ignored, leading to incorrect behaviour in JSC.

jacobbramley, 2023-12-14:
_Originally posted by @brooksdavis in https://github.com/CTSRD-CHERI/cheribsd/issues/1964#issuecomment-1852503589_:
> With mrs we're just ignoring the TCACHE and ARENA flags. We could probably pass them through, but I'm not convinced they would be particularly meaningful given the effects to having a single, arbitrarily selected thread insert everything into the free list on easy cycle. There's defiantly more work to do here on micro optimization in a quarantined environment.

Sadly, `MALLOCX_ARENA` isn't just used as an optimisation here. JavaScriptCore [defines an arena](https://github.com/capablevms/webkit/blob/44f2044321e7500b8aef8ad3a015b6c1c68522d7/Source/WTF/wtf/ContinuousArenaMalloc.cpp#L89), [sets `DDC` to cover it](https://github.com/capablevms/webkit/blob/44f2044321e7500b8aef8ad3a015b6c1c68522d7/Source/WTF/wtf/ContinuousArenaMalloc.cpp#L100), then [uses `MALLOCX_ARENA`](https://github.com/capablevms/webkit/blob/44f2044321e7500b8aef8ad3a015b6c1c68522d7/Source/WTF/wtf/ContinuousArenaMalloc.cpp#L114) to allocate pointers that can be used with that `DDC` (e.g. in generated code).

The JSC Morello port supports a compressed-pointer mode, and this mechanism is likely to be important there, but this bug actually occurs on a standard purecap build. I'm not sure why we're using plain pointers in such a build, but it does mean that we'll see a regression until we can investigate.

For now, we'll disable revocation when we test JSC, but I thought it'd be useful to report the issue anyway.

jrtc27, 2023-12-14:
At that point you really just need your own allocator. Other allocators will never support that, and I hope we can remove the jemalloc interfaces entirely from FreeBSD in future.

I guess we should make mrs reject allocations with some of those flags set, but that won’t be in our 23.11 release.

brooksdavis, 2023-12-14:
The easiest answer here is likely to link a private copy of jemalloc with suitable renaming (supported in the codebase already) and use that for these allocations. They shouldn't really be subject to revocation in the normal way, probably just when the arena is destroyed and the OS takes care of it by revoking all capabilities to the now quarantined reservation.