CTSRD-CHERI/cheribsd #1989: bhyve gdb stub for Morello

bsdjhb, 2024-01-31:


markjdb, 2024-01-31:
Currently waiting for some commits to land upstream so that I can avoid creating unnecessary divergence in CheriBSD:
https://reviews.freebsd.org/D43664
https://reviews.freebsd.org/D43665
https://reviews.freebsd.org/D43666

markjdb, 2024-01-31:
My WIP branch for CheriBSD is here: https://github.com/markjdb/cheribsd/tree/dev-bhyve-gdb3

It reverts the previous gdb stub patches in favour of something more complete. Also because the gdb stub changed upstream and it was simpler to revert and then backport the upstream changes. Once the aforementioned differential revisions land upstream in some form, I'll open a PR.

bsdjhb, 2024-04-17:
These have all landed upstream now it looks like.

brooksdavis, 2024-06-04:
Are we still waiting on fixes for -G or is this about ready to land?

markjdb, 2024-06-04:
> Are we still waiting on fixes for -G or is this about ready to land?

As it happens, I'm finally about to push these patches into FreeBSD (Andy approved them last night), so I'll re-do the CheriBSD PR with upstream cherry-picks. Should be ready within a day.

closed 2024-08-14 by markjdb