CTSRD-CHERI/cheribsd #1996: mrs: enable worker thread

brooksdavis, 2024-02-02:
Possible stopgap before switching to a kernel thread to make cheri_revoke non-blocking.

brooksdavis, 2024-02-07:
This isn't as simple as defining OFFLOAD_QUARANTINE because that is based on pthreads and we can't easily add a pthreads dependency to libc. We either need to roll something using thr_new() and _umtx_op() or add a kernel worker thread.

markjdb, 2024-02-14:
I've got a simple implementation of a `CHERI_REVOKE_ASYNC` flag. The basic idea is that the cheri_revoke(2) caller will stop threads in the process and revoke from horders before signaling the kernel worker thread to scan the vmspace. In particular, I don't see a good reason to make all of cheri_revoke() asynchronous: it seems silly to signal a worker thread only to have it turn around and suspend the signaling process.

I'm currently looking at MRS and OFFLOAD_QUARANTINE, since I'm not familiar with them. In general there's still a fair bit of work that needs to happen in userspace, to clear the bit map and free slabs to the underlying allocator. Do we have a sense for whether this needs to be asynchronous as well, at least for an initial implementation?

markjdb, 2024-03-16:
Now that async support in the kernel has landed, most of this is done, but we still update bitmaps and drain quarantine arenas in the critical path. We might still want a dedicated thread to handle this, or implement some work-limiting mechanism to try and ensure that application threads don't observe too much latency.

gvnn3, 2024-05-01:
Closing as we will be doing much of the rest of this as research elsewhere.

closed 2024-05-01 by gvnn3