CTSRD-CHERI/cheribsd #2034: mmap() can't be called via (__)syscall

arichardson, 2024-03-07:
This is used e.g. by the sanitizer runtimes, but since unistd.h declares syscall/__syscall with an integer return value it's not possible to actually call mmap() successfully:

https://github.com/CTSRD-CHERI/cheribsd/blob/698d1636dd1fe2322e5bc7029e415928c80b76b1/include/unistd.h#L582-L583

Should these be changed to intptr_t for purecap?

brooksdavis, 2024-03-07:
I suspect we should leave syscall alone (it can't be used for mmap except by accident), but likely __syscall should be int64ptr_t.

jrtc27, 2024-03-07:
Using it for mmap has two problems:
1. off_t is 64-bit, so on a 32-bit system you need to use __syscall so the arguments are aligned correctly
2. Pointers get truncated on a 64-bit system even absent CHERI

There's also shmat which suffers from 2 but not 1.