CTSRD-CHERI/cheribsd #2035: kernel panic on `service netif restart` with jails + ssh

dstolfa, 2024-03-07:
I'm unsure what actually caused this panic, but the situation was that I had a cloned interface `lo1` in `/etc/rc.conf` and issued a `service netif restart` while ssh'd to see if things work. Attached at the bottom is the backtrace. I wasn't able to get a core dump this time since I didn't have enough swap space, but should I successfully reproduce it I'll attach that as well.

```
login: Mar  4 21:03:24 morello26 sshd[71204]: error: PAM: Authentication error for illegal user squid from 45.143.201.62
x0: 0xffff000193be3fa8 [rwRW,0xffff000193be3f98-0xffff000193be4020]
x1: 0xffff000193be3f98 [rwRW,0xffff000193be3f98-0xffff000193be4020]
x2: 0xffffa0001070c000 [rwRW,0xffffa0001070c000-0xffffa0001070d000]
x3: 0xffffa08de69f8600 [rwRW,0xffffa08de69f8600-0xffffa08de69f8780]
x4: 0xffff0000007faead [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (nlattr_get_uint32 + 0x0)
x5: 0xffff000193be3eb0 [rwRW,0xffff000193be3eb0-0xffff000193be3f30]                                                                                                                             x6: 0xffff000000b34e69 [rR,0xffff000000b34e69-0xffff000000b34e8f] (aarch32_esigcode + 0xba2a1)
x7: 0xffff000000a8b7d7 [rR,0xffff000000a8b7d7-0xffff000000a8b7e7] (aarch32_esigcode + 0x10c0f)
x8: 0x0000000000002000
x9: 0x0000000000002405                                                                                                                                                                          x10: 0x0100000000000000
x11: 0x0000000000000000
x12: 0x0000000000001c1c
x13: 0x0000000000000008
x14: 0x0000000000000046
x15: 0xffff000000b3ae7c [rR,0xffff000000b3ae7c-0xffff000000b3ae8f] (aarch32_esigcode + 0xc02b4)
x16: 0xffff000000ae5c63 [rR,0xffff000000ae5c63-0xffff000000ae5ca7] (aarch32_esigcode + 0x6b09b)
x17: 0x0000000000000044
x18: 0xffff000193be3e00 [rwRW,0xffff000193bdf000-0xffff000193be5000]
x19: 0xffff000193be3f98 [rwRW,0xffff000193be3f98-0xffff000193be4020]
x20: 0xffffa08de7280800 [rwRW,0xffffa08de7280800-0xffffa08de7280a00]
x21: 0xffffa0001070c000 [rwRW,0xffffa0001070c000-0xffffa0001070d000]
x22: 0x0000000081206919
x23: 0xffffa08de69f8600 [rwRW,0xffffa08de69f8600-0xffffa08de69f8780]
x24: 0xffff000193be3fa8 [rwRW,0xffff000193be3fa8-0xffff000193be40b8] (invalid)
x25: 0xffffa080b5a07000 [rwRW,0xffffa080b5a07000-0xffffa080b5a07800]
x26: 0x0000000000000000
x27: 0x0000000000000000
x28: 0xffff0000008029f9 [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (rtnl_handle_message + 0x0)
x29: 0xffff000193be3e00 [rwRW,0xffff000193bdf000-0xffff000193be5000]
ddc: 0x0000000000000000
sp: 0xffff000193be3e00 [rwRW,0xffff000193bdf000-0xffff000193be5000]
lr: 0xffff00000080607d [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (rtnl_handle_addr + 0x4bc)
elr: 0xffff0000007a3f00 [rxR,0x0000000000000000-0xffffffffffffffff] (in6_control_ioctl + 0x1ef)
spsr: 0x0000000004400045
far: ffff000193be3fa9
esr:         96000028                                                                                                                                                                           panic: Capability abort from kernel space: tag violation                                                                                                                                        cpuid = 3                                                                                                                                                                                       time = 1709587207                                                                                                                                                                               KDB: stack backtrace:                                                                                                                                                                           db_trace_self() at db_trace_self                                                                                                                                                                db_trace_self_wrapper() at db_trace_self_wrapper+0x48                                                                                                                                           vpanic() at vpanic+0x1ec                                                                                                                                                                        panic() at panic+0x4c                                                                                                                                                                           cap_abort() at cap_abort+0x24c                                                                                                                                                                  handle_el1h_sync() at handle_el1h_sync+0x14                                                                                                                                                     --- exception, esr 0x96000028                                                                                                                                                                   in6_control_ioctl() at in6_control_ioctl+0x1f0
rtnl_handle_addr() at rtnl_handle_addr+0x4b8                                                                                                                                                    rtnl_handle_message() at rtnl_handle_message+0x244                                                                                                                                              nl_taskqueue_handler() at nl_taskqueue_handler+0x818                                                                                                                                            taskqueue_run_locked() at taskqueue_run_locked+0x180                                                                                                                                            taskqueue_thread_loop() at taskqueue_thread_loop+0xf0                                                                                                                                           fork_exit() at fork_exit+0x8c                                                                                                                                                                   fork_trampoline() at fork_trampoline+0x14p                                                                                                                                                      KDB: enter: panic                                                                                                                                                                               [ thread pid 0 tid 101699 ]                                                                                                                                                                     Stopped at      kdb_enter+0x5f: undefined       c200027f                                                                                                                                                                                                                                                                                        ```

bsdjhb, 2024-03-08:
Can you open gdb on the kernel executable and do `l *in6_control_ioctl+0x1f0` well as `x/-10i in6_control_ioctl+0x1f0`?

dstolfa, 2024-03-08:
```
(gdb) l *in6_control_ioctl+0x1f0
0xffff0000006fd1e0 is in in6_control_ioctl (/home/ds815/cheribsd-dev/sys/netinet6/in6.c:412).
407                     sa6 = NULL;
408                     break;
409             }
410             if (sa6 && sa6->sin6_family == AF_INET6) {
411                     if (sa6->sin6_scope_id != 0)
412                             error = sa6_embedscope(sa6, 0);
413                     else
414                             error = in6_setscope(&sa6->sin6_addr, ifp, NULL);
415                     if (error != 0)
416                             return (error);
(gdb) x/-10i in6_control_ioctl+0x1f0
   0xffff0000006fd1b8 <in6_control_ioctl+456>:  lsl     x9, x9, x8
   0xffff0000006fd1bc <in6_control_ioctl+460>:  movk    x10, #0xd, lsl #48
   0xffff0000006fd1c0 <in6_control_ioctl+464>:  tst     x9, x10
   0xffff0000006fd1c4 <in6_control_ioctl+468>:  b.eq    0xffff0000006fd2a4 <in6_control_ioctl+692>  // b.none
   0xffff0000006fd1c8 <in6_control_ioctl+472>:  ldrb    w8, [x19, #17]
   0xffff0000006fd1cc <in6_control_ioctl+476>:  add     x24, x19, #0x10
   0xffff0000006fd1d0 <in6_control_ioctl+480>:  cmp     w8, #0x1c
   0xffff0000006fd1d4 <in6_control_ioctl+484>:  b.ne    0xffff0000006fd238 <in6_control_ioctl+584>  // b.any
   0xffff0000006fd1d8 <in6_control_ioctl+488>:  ldr     w8, [x19, #40]
   0xffff0000006fd1dc <in6_control_ioctl+492>:  cbz     w8, 0xffff0000006fd338 <in6_control_ioctl+840>
   ```