CTSRD-CHERI/cheribsd #2037: Git times out when quarantining is enabled in CheriBSD 23.11

rwatson, 2024-03-10:
I with extreme reliability get a `git` clone failure on large repositories with quarantining enabled in the out-of-the-box CheriBSD 23.11 release on my Morello desktop machine:

```
# git clone https://github.com/CTSRD-CHERI/cheribsd
Cloning into 'cheribsd'...
remote: Enumerating objects: 4256953, done.
remote: Counting objects: 100% (56702/56702), done.
remote: Compressing objects: 100% (22550/22550), done.
error: 7868 bytes of body are still expected6.37 MiB | 1.24 MiB/s   
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
```

This appears to be due to long synchronous pause times (as observed by @jrtc27).

rwatson, 2024-03-10:
Running with kernel async revocation (#2011) appears to reliably resolve this issue.

rwatson, 2024-03-10:
@gvnn3 something we strongly want resolved in the next release / for late March 2024 Morello demonstrations.

rwatson, 2024-03-18:
This is resolved with #2011 merged, and if it is enabled. To fully resolve this issue, we need to enable asynchronous revocation by default; perhaps @markjdb can comment on the wisdom of such a choice for 24.04?

markjdb, 2024-03-19:
Closing since the default has been flipped in `dev` by https://github.com/CTSRD-CHERI/cheribsd/pull/2058

closed 2024-03-19 by markjdb