CTSRD-CHERI/cheribsd #2131: sort on a large file never ends with enabled revocations

kwitaszczyk, 2024-06-27:
On dev (https://github.com/CTSRD-CHERI/cheribsd/commit/bdeff30fb6b1744816f43ed8a3c2f0a133d872c1) running GENERIC-MORELLO-PURECAP, I executed `make cscope CSCOPE_ARCHDIR=arm64` in sys/ and I noticed it never finishes with `security.cheri.runtime_revocation_default` set to `1`, even though `cscope` was hybrid.

It turned out it was a CheriABI `sort` process that doesn't finish:
```
kw543@cheribsd:~ $ sudo sysctl security.cheri.runtime_revocation_default=0
security.cheri.runtime_revocation_default: 1 -> 0
kw543@cheribsd:~ $ time env LC_ALL=C sort -T /tmp/ cscope.1548/cscope.1 >/dev/null
       23.75 real        21.85 user         1.89 sys
kw543@cheribsd:~ $ sudo sysctl security.cheri.runtime_revocation_default=1
security.cheri.runtime_revocation_default: 0 -> 1
kw543@cheribsd:~ $ time env LC_ALL=C sort -T /tmp/ cscope.1548/cscope.1 >/dev/null
load: 1.50  cmd: sort 1667 [running] 227.85r 39.12u 102.12s 47% 18137712k
```

markjdb, 2024-06-27:
`sort`'s got several realloc loops which grow the allocation by a fixed number of entries. cscope generates a decently large input file (230MB), so `sort` calls realloc() many times. This interacts poorly with MRS, which implements realloc() by always allocating a new buffer, copying, and freeing the old one. In particular, the quarantine ends up growing very quickly and triggers many revocation scans.

This behaviour in `sort` dates back to the original implementation, I guess it's just fast enough that no one's really tried to tune it. Changing all of the loops to grow the allocation size by a factor of 2 rather than just adding 1024 or 128 new entries lets `sort` finish in a reasonable amount of time. With that change on my morello system, a `make -C sys cscope CSCOPE_ARCHDIR=arm64` takes:
- 43s with revocation disabled
- 1m3s with async revocation enabled
- 1m17s with synchronous revocation enabled

We should try to find out how much of this overhead is caused by MRS having a pessimal realloc() implementation.

For this release, I suppose we could simply commit these changes to `sort`. I think they are low-risk.

markjdb, 2024-06-27:
A couple of experiments to try when I have a bit more time (or if someone else would like to take a look):
- How much of this overhead comes just from MRS allocating a new buffer and copying the old buffer?
- @brooksdavis notes that mrs_realloc() might be able to trivially satisfy a realloc() of a large buffer if the capability bounds have been rounded up for representability. Checking for that might help avoid blowups of other "conservative" reallocation loops.

brooksdavis, 2024-06-27:
I think it's broken that sort is O(n^2) when realloc doesn't mask it by extending in place. There's no good reason for that (I guess virtual memory size limits, but we normally we live in an overcommit world were address space is free.)

markjdb, 2024-06-27:
> I think it's broken that sort is O(n^2) when realloc doesn't mask it by extending in place. There's no good reason for that (I guess virtual memory size limits, but we normally we live in an overcommit world were address space is free.)

I'm 99% sure there's no principled reason for doing it this way. I'll propose changing that upstream.

markjdb, 2024-06-27:
> I'm 99% sure there's no principled reason for doing it this way. I'll propose changing that upstream.

We could always fall back to a more conservative behaviour if we detect a virtual memory limit, but I'm not sure it's worth the hassle until we see a reason. I've only ever seen that limit used to try and catch memory leaks before they consume all of a system's RAM.

markjdb, 2024-07-17:
The current status is:
- the problem is mitigated by #2137 
- the combination of #2133 and #2147 make the mitigation unnecessary.

bsdjhb, 2025-03-31:
Can we close this now?  The only thing I can think of is if we want to revert #2137 now that we've mitigated in other ways (or if we want to instead upstream #2137)?

brooksdavis, 2025-03-31:
I tend to think we should upstream removing the O(n^2) behavior from sort (I think having the code written that way encourages a delusion that memory/address space is being saved), but closing this seems fine.

closed 2025-03-31 by brooksdavis