CTSRD-CHERI/cheribsd #2253: Purecap nvi(1) crash

rwatson, 2024-11-27:
I was casually editing a kernel source-code file on my Morello box, running with pure cap + heap temporal safety + c18n and encountered this crash:

```
Core was generated by `vi kern_proc.c'.
Program terminated with signal SIGPROT, CHERI protection violation.
Capability bounds fault.
#0  apply_with (sp=0x41410400 [rwRW,0x41410400-0x41411730], lno=2620, 
    p=0x4299af91 [rwRW,0x42993000-0x4299b060] "\t", len=<optimized out>, 
    db_func=<optimized out>) at /usr/src/contrib/nvi/common/log.c:740
740                             bp[i] = (lp[i] >> offl) ^ (lp[i+1] << offr);
(gdb) bt
#0  apply_with (sp=0x41410400 [rwRW,0x41410400-0x41411730], lno=2620, 
    p=0x4299af91 [rwRW,0x42993000-0x4299b060] "\t", len=<optimized out>, 
    db_func=<optimized out>) at /usr/src/contrib/nvi/common/log.c:740
#1  log_backward (sp=0x41410400 [rwRW,0x41410400-0x41411730], 
    rp=<optimized out>) at /usr/src/contrib/nvi/common/log.c:435
#2  0x0000000000169620 in vi (spp=0x41fffda0 [rwRW,0x41fffda0-0x41fffdb0])
    at /usr/src/contrib/nvi/vi/vi.c:226
#3  0x000000000013b498 in editor (gp=0x4140f000 [rwRW,0x4140f000-0x414102e0], 
    argc=<optimized out>, argv=<optimized out>)
    at /usr/src/contrib/nvi/common/main.c:402
#4  0x0000000000130d0c in main (argc=2620, 
    argv=0xffffbff7f540 [rwRW,0xffffbff7f540-0xffffbff7f570])
    at /usr/src/contrib/nvi/cl/cl_main.c:115
(gdb) info registers
x0             0x415058c8          1095784648
x1             0x4299b060          1117368416
x2             0x415058d0          1095784656
x3             0x0                 0
x4             0x41fff3c0          1107293120
x5             0x414f8030          1095729200
x6             0x0                 0
x7             0x0                 0
x8             0x8                 8
x9             0x38                56
x10            0x1                 1
x11            0x2900000029        176093659177
x12            0x0                 0
x13            0x1f                31
x14            0xcc                204
x15            0x1                 1
x16            0x4055d820          1079367712
x17            0x403f0b8d          1077873549
x18            0x0                 0
x19            0x41410400          1094779904
x20            0x0                 0
x21            0xa3c               2620
x22            0x41fff720          1107293984
x23            0x1a4f70            1724272
x24            0x1172b1            1143473
x25            0x4299af91          1117368209
x26            0x41fff6d8          1107293912
x27            0x41481200          1095242240
x28            0x4299af89          1117368201
x29            0x41fff410          1107293200
x30            0x139e3d            1285693
sp             0xfffffff80000      281474976186368
pc             0x13a124            1286436
cpsr           0x24000200          [ EL=0 D BTYPE=0 C64 C ]
fpsr           0x0                 [ ]
fpcr           0x0                 [ Len=0 Stride=0 RMode=0 ]
c0             0xdc5d40005a00580000000000415058c8 0x415058c8 [rwRW,0x41505800-0x41505a00]
c1             0xdc5d400018379806000000004299b060 0x4299b060 [rwRW,0x42993000-0x4299b060]
c2             0xdc5d40005a00580000000000415058d0 0x415058d0 [rwRW,0x41505800-0x41505a00]
c3             0x0                 0x0
c4             0xdc5f400073e0f3c00000000041fff3c0 0x41fff3c0 [rwRW,0x41fff3c0-0x41fff3e0]
c5             0xdc5d40000037c00600000000414f8030 0x414f8030 [rwRW,0x414f8000-0x41500060]
c6             0x0                 0x0
c7             0x0                 0x0
c8             0x8                 0x8
[... 94 lines cut ...]
It seems that `c1` (possibly `lp`?) is at the end of its buffer and attempting to load pair (`ldp`) from 8 bytes short of that, which isn’t enough for 64-bit registers let alone 128-bit ones:

```
c1             0xdc5d400018379806000000004299b060 0x4299b060 [rwRW,0x42993000-0x4299b060]
```

The context here is:
```
735                     }
736                     for (i = 0; i < cnt; ++i)
737     #if BYTE_ORDER == BIG_ENDIAN
738                             bp[i] = (lp[i] << offl) ^ (lp[i+1] >> offr);
739     #else
740                             bp[i] = (lp[i] >> offl) ^ (lp[i+1] << offr);
741     #endif
```

`lp` is of type `nword *`, which is in fact `unsigned long *`, and it isn’t surprising that the compiler might (via SROA or similar) try to convert the two (statically) adjacent 8-byte reads into a wider read. But there appear to be two problems here:

 1. `ldp` in pure cap code will load two adjacent 16-byte values, not two adjacent 8-byte values.
 2. Even if it were a ldp for two adjacent 8-byte reads .. they still wouldn’t fit in the remaining buffer space, as the pointer is to the upper bound location, and the offset is -8 and not -16.

I haven’t tried to plumb the code generation further, but it seems like at the very least `ldp` is the wrong thing here, and no matter how we look at it, the immediate offset is likely wrong. But a more detailed reading of the code generation is required to understand this better, which I don’t have time for currently.

I have the binary and core dump easily to hand [for now].

jrtc27, 2024-11-27:
Is this https://github.com/lichray/nvi2/pull/122/files not having been backported?

brooksdavis, 2024-11-27:
> Is this https://github.com/lichray/nvi2/pull/122/files not having been backported?

Looks like lichray hasn't pushed out a new release since just before the fix. I'll see about looping the fix through FreeBSD.

brooksdavis, 2024-12-03:
Fixed in dev so closing

PR #2254 (2024-12-03): nvi fixes
Merge some fixes from upstream nvi via FreeBSD. This was the only bit of code in nvi that was changed to support CHERI and a the commit messages don't really do justice to how utterly bonkers the code was.

Fixes #2253
closed 2024-12-03 by brooksdavis