CTSRD-CHERI/cheribsd #2256: Unneeded CHERI subobject bounds for copying link-level socket address when IP_RECVIF socket option used

RoundofThree, 2024-12-03:
Syzkaller found the following crash in a purecap kernel with subobject bounds:
```
  x0: 0xffff00013237d0c6 [rwxRW,0xffff00013237d090-0xffff00013237d0c6]
  x1: 0xffff008000000000 [rwRW,0xffff008000000000-0xffff00a000000000]
  x2: 0xffff007ffd800000 [rwxRW,0xffff000000000000-0xffff008000000000]
  x3: 0xffff000001282825 (_memcpy + 0x304)
  x4: 0xffff00014acbb840 [rwRW,0xffff00014acbb840-0xffff00014acbb844]
  x5: 0xffff00014acbb800 [rwRW,0xffff00014acbb800-0xffff00014acbb880]
  x6: 0x0000000000000001
  x7: 0xffff00013237fa50 [rwxRW,0xffff000132370000-0xffff000132380000]
  x8: 0x0000000000000000
  x9: 0x0000000000040000
 x10: 0x00000000000001ac
 x11: 0x0000000000000000
 x12: 0x0000000000000007
 x13: 0x0000000000000007
 x14: 0x0000000000000060
 x15: 0x0000000000000040
 x16: 0xffff000002cecf70 [rxR,0xffff000002c9b000-0xffff000002cf5e00] (__stop_set_modmetadata_set + 0x650)
 x17: 0xffff000000afa5b5 [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (__sanitizer_cov_trace_pc + 0x0)
 x18: 0xffff00013237ce50 [rwxRW,0xffff000132370000-0xffff000132380000]
 x19: 0xffff00013237d090 [rwxRW,0xffff00013237d090-0xffff00013237d0c6]
 x20: 0x0000000000000001
 x21: 0xffff0001e1c7e287 [rwRW,0xffff0001e1c7e180-0xffff0001e1c7e300]
 x22: 0x0000000000000000
 x23: 0xffff00013237d0c6 [rwxRW,0xffff00013237d090-0xffff00013237d0c6]
 x24: 0xffff00013237d0c0 [rwxRW,0xffff00013237d090-0xffff00013237d0c6]
 x25: 0x0000000000000000
 x26: 0x0000000000000000
 x27: 0x0000000000000038
 x28: 0x0000000000000001
 x29: 0xffff00013237ce60 [rwxRW,0xffff000132370000-0xffff000132380000]
 ddc: 0x0000000000000000
  sp: 0xffff00013237ce50 [rwxRW,0xffff000132370000-0xffff000132380000]
  lr: 0xffff000001282841 [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (_memcpy + 0x320)
 elr: 0xffff000001282840 [rxR,0x0000000000000000-0xffffffffffffffff] (_memcpy + 0x31f)
spsr: 0x0000000084400045
 far: ffff00013237d0c6
 esr:         9600006a
panic: Capability abort from kernel space: bounds violation
cpuid = 0
time = 1733228807
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_trace_self_wrapper() at db_trace_self_wrapper+0xd4
kdb_backtrace() at kdb_backtrace+0xc8
vpanic() at vpanic+0x2b4
panic() at panic+0xc4
cap_abort() at cap_abort+0x3e8
handle_el1h_sync() at handle_el1h_sync+0x14
--- exception, esr 0x9600006a
_memcpy() at _memcpy+0x320
ip_savecontrol() at ip_savecontrol+0xd54
rip_append() at rip_append+0x250
rip_input() at rip_input+0x5d0
icmp_input() at icmp_input+0x37c
ip_input() at ip_input+0x750
netisr_dispatch_src() at netisr_dispatch_src+0x284
ether_demux() at ether_demux+0x3ac
ether_nh_input() at ether_nh_input+0x988
netisr_dispatch_src() at netisr_dispatch_src+0x284
ether_input() at ether_input+0x1c8
tunwrite() at tunwrite+0xd4c
devfs_write_f() at devfs_write_f+0x24c
dofilewrite() at dofilewrite+0x10c
kern_writev() at kern_writev+0xe0
sys_write() at sys_write+0x1b0
do_el0_sync() at do_el0_sync+0xf84
handle_el0_sync() at handle_el0_sync+0x34
--- exception, esr 0x56000000
KDB: enter: panic
[ thread pid 3131 tid 100773 ]
Stopped at      kdb_enter+0x8f: undefined       c20002bf
db> 
```

The crash is triggered in this line of code:
https://github.com/CTSRD-CHERI/cheribsd/blob/6fb9ad460cd6b24b3a37c09c0aefe1de39600879/sys/netinet/ip_input.c#L1246
because `sdlbuf` accounts for any possible extra data that follows after `struct sockaddr_dl` in `sdp = (struct sockaddr_dl *)ifp->if_addr->ifa_addr`. 
```c
/* OMITTED */
		struct sdlbuf {
			struct sockaddr_dl sdl;
			u_char	pad[32];  // padding to account for extra data
		} sdlbuf;
		struct sockaddr_dl *sdp;
		struct sockaddr_dl *sdl2 = &sdlbuf.sdl;
		if ((ifp = m->m_pkthdr.rcvif)) {
			sdp = (struct sockaddr_dl *)ifp->if_addr->ifa_addr;
/* OMITTED */
```
However, if subobject bounds is enabled, the compiler will bound the `sdl2` capability to `sizeof(struct sockaddr_dl)` due to https://github.com/CTSRD-CHERI/cheribsd/blob/6fb9ad460cd6b24b3a37c09c0aefe1de39600879/sys/netinet/ip_input.c#L1235
Therefore, if `sdp->sdl_len > sizeof(struct sockaddr_dl)`, the  memory copy will lead to a capability bounds fault in https://github.com/CTSRD-CHERI/cheribsd/blob/6fb9ad460cd6b24b3a37c09c0aefe1de39600879/sys/netinet/ip_input.c#L1246 

The crash can be triggered by a local unprivileged user in CheriBSD. 

Tagging @YiChenChai

bsdjhb, 2024-12-03:
The `sdl` member of `sdlbuf` just needs to be tagged with `__subobject_use_container_bounds` to fix this.

RoundofThree, 2024-12-03:
Yes.

PR #2258 (2024-12-05): cheri: Use container bounds when accessing sockaddr_dl from sdlbuf
Fixes #2256.
commit 0d253640 (2024-12-05): cheri: Use container bounds when accessing sockaddr_dl from sdlbuf

Fixes #2256.