CTSRD-CHERI/cheribsd #2355: bhyve guest crashes when stepping over a branching instruction

kwitaszczyk, 2025-03-05:
A bhyve VM running [kernel-c18n](https://github.com/CTSRD-CHERI/cheribsd/tree/kernel-c18n) crashes when setting a breakpoint after a `blr` instruction and continuing execution in gdb. The kernel switches between the Restricted and Executive modes that could affect bhyve here.

Steps to reproduce:
1. Run the disk image of kernel-c18n under bhyve
2. Attach GDB to the VM
3. In GDB:
    ```
    b compressor_init
    ```
    ```
    c
    ```
4. In bhyve:
    ```
    kldload zlib_c18n
    ```
    ```
    zlibtest
    ```
5. GDB should hit the breakpoint at this stage. Step through the kernel with `ni` until reaching the first `blr` (e.g., `blr c4`)
6. In GDB:
    ```
    b *($pcc+4)
    ``` 
    ```
    c
    ``` 
7. The guest should panic at this stage with a message like:
    ```
    Stopped at      compressor_init+0x5b:   undefined       b4000295
    ```

Host:
```
FreeBSD stevnsbaer 15.0-CURRENT FreeBSD 15.0-CURRENT #2 dev-n274330-514fcb667376: Tue Feb 18 16:54:41 GMT 2025     root@stevnsbaer:/usr/obj/usr/src/arm64.aarch64c/sys/GENERIC-MORELLO-PURECAP arm64
```
https://github.com/CTSRD-CHERI/cheribsd/pull/2255 and https://github.com/CTSRD-CHERI/cheribsd/pull/2329 were applied before building this kernel.

`gdb-cheri` package: 14.1.d20250221_1

bsdjhb, 2025-03-05:
Does this work correctly if you run the guest image under QEMU instead of bhyve?

kwitaszczyk, 2025-03-05:
> Does this work correctly if you run the guest image under QEMU instead of bhyve?

Yes, the same scenario works as expected with a VM under QEMU.

kwitaszczyk, 2026-01-09:
#2483 seems to resolve this issue. Thank you for working on that!

closed 2026-01-09 by kwitaszczyk