CTSRD-CHERI/cheribsd #2446: Fix `je_malloc_underlying_allocation` to correctly validate sub-capabilities

MauriceDHanisch, 2025-08-27:
### Description:

While investigating `je_malloc_underlying_allocation` we found that its current implementation relies too heavily on the input pointer (as noted in existing comments). It only checks that the offset is non-zero and then proceeds, which means a sub-capability derived from a valid allocation with adapted bounds can be passed to free(). This undermines heap spatial safety (as described in [Brett Gutstein’s dissertation](https://www.cl.cam.ac.uk/techreports/UCAM-CL-TR-975.pdf)). It can also be used to break temporal safety, since freeing a sub-capability causes the quarantine logic to treat the wrong region as freed, unpainting shadow bits for addresses that still belong to live allocations.

There seems to be two possible design choices:

1. Accept subcapabilities:
Treat any capability derived from an allocation as valid, look up the owning extent, and return a capability to the underlying allocation. This ensures free/underlying_allocation works even if passed a subcapability.
    - We implemented this in branch [get-underlying-allocation](https://github.com/MauriceDHanisch/cheribsd/tree/get-underlying-allocation).
    - This version validates the capability (tag, bounds, seal bit), then uses the rtree and size class arithmetic to compute the correct region base inside the extent.

2. Reject subcapabilities:
Enforce that only the original allocation capability may be passed. If a subcapability is provided, the allocator traps with an error.
    - We implemented this in branch [check-underlying-allocation](https://github.com/MauriceDHanisch/cheribsd/tree/check-underlying-allocation).
    - This version validates the capability and explicitly aborts if it does not point to the base of an allocation region.

### Testing:
Both branches were tested with the CheriBSD lib/libc/stdlib Kyua suite: `kyua test -k /usr/tests/Kyuafile lib/libc/stdlib`. Results were unchanged compared to the current tree: all tests passed except posix_memalign_test:aligned_alloc_basic and posix_memalign_basic.

### Discussion point:
It’s unclear what the intended contract is: should the allocator be permissive (resolve subcaps) or strict (trap on non-base capabilities)? We would appreciate guidance on which direction is preferred before turning this into a PR.

MauriceDHanisch, 2025-08-28:
> We generally take the view that the C standard is clear that the capability passed to free must be exactly the one returned by malloc [...]

Okay!

> There's a case to be made that the correct fix is a mix of the two solutions you've proposed: use rtree, etc. to create a capability that should be the one malloc handed out and then use cheri_equal_exact to compare it to to the passed capability.

The problem is that you can only find out the size class in which the requested size is/was located, but not the requested size itself. In addition, SW_VMEM is stripped, so you need to take this into account. In practical terms, you can only check the permissions (without SW_VMEM) and the start address.

closed 2025-08-28 by MauriceDHanisch

brooksdavis, 2025-08-28:
> Wouldn't `mrs_bound_pointer` bound it to the requested size?

It would, but we could remove that functionality if it got it the way.